verilator/verilator #5378: %Warning-REALCVT with variable of type time

RootCubed, 2024-08-16:
Attempting to verilate the following code on master (``verilator --binary --exe --build -j `nproc` timecvt.sv -o timecvt``):
```sv
`timescale 1ns/1ps

time T = 9ns;

module top(); endmodule
```

outputs a warning:
```
%Warning-REALCVT: timecvt.sv:3:10: Implicit conversion of real to integer
    3 | time T = 9ns;
      |          ^~~
                  ... For warning description see https://verilator.org/warn/REALCVT?v=5.027
                  ... Use "/* verilator lint_off REALCVT */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Interestingly, doing one of the following things all cause the warning to go away:
- Removing the line `timescale 1ns/1ps`
- Changing the constant to e.g. `8ns`
- Removing the unit from the constant, so `time T = 9;`

I haven't looked into what the cause of this might be, but it sounds like some kind of floating-point rounding issue, possibly?

wsnyder, 2024-08-18:
Yes, the problem is the floating point loss of precsion times 1000 is more then epsilon (the floating point error).  I suppose we need to check specially for this warning before we do the time scale conversion, then suppress the warning for any later checks (after the factor of 1000 correction).

Might you want to make a pull request to fix this?

RootCubed, 2024-08-19:
So I just realized that the `time` datatype is actually just an integer type with 64 bits... using `parameter` instead works just fine.

I think the problem for this issue is that the rounding error happens very early on, namely during parsing. The `_001_cells.tree` already has `1:2:3:1: CONST 0x55a0379f2eb0 {d3ap} @dt=0x55a0379f1a70@(G/d)  8.9999999999999995e-09`, so it seems tricky to tell when issuing a rounding warning is warranted and when not (we probably want to show REALCVT for `T = 9.0000001ns` but not for `T = 9ns`.

closed 2024-08-24 by wsnyder