verilator/verilator #5379: Terrible trace performance on huge design

jcteng, 2024-08-16:
Thanks for taking the time to report this.

What would you like added/supported?
Improve FST tracing performance

What 'verilator --version' are you using?  Did you try it with the git master version?
5.027

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
Yes,  I can upload project  for reprodurce after few days

May we assist you in trying to fix this in Verilator yourself?

TargetHDL：
RTL generated from the 2Cores configuration of the Xiangshan project 
-----------------------------

Verilator achieves parallelization in the non-blocking assignment (NBA) scheduling of Verilog, resulting in significant simulation performance improvements compared to other current commercial products. With proper RTL design, it can achieve simulation speeds over 10x faster than single-threaded simulations.

However, with fst trace enabled, especially in the case of large footprint executable files, the simulation performance quickly drops below single-threaded performance, with a 10x difference in shock resistance compared to when fst tracing is disabled. To investigate the cause of this phenomenon, the RTL generated from the 2Cores Config of the Xiangshan project was used as a sample for analysis.

In the XiangShan2Cores project, TraceDump utilized over 70-90% of the CPU time in the main thread (the figure below shows one scenario at 70%), excessively occupying the main thread, which led to a 3x reduction in the evaluation efficiency of the main thread.
[image]

The main reason for taking too long is icache miss. There are a huge number of functions that collect signals, and the execution of each function has different degrees of frontend bound. Some of them have icache missing up to 100%, and the average level is above 50%. This undoubtedly limits the performance of the CPU. It is very frustrating to have such high icache missing in a single task such as signal acquisition. It is recommended to refactor this part of the code. The generation of C++ code in Emitter can consider better expression forms and centralized processing of signals. This can avoid low icache performance caused by traces being scattered in different locations.
[image]

wsnyder, 2024-08-17:
Regardless of improving tracing and choice of simulator, note the best for performance is always to trace for the minimum amount of time needed, and only trace the signals needed.

The best for performance would be to implement GTKwave (gtkwave/gtkwave#286), then have Verilator support this. Then your e.g. 10 threads dump data in parallel, from your notes this might give 5x speed up (if your I/O is fast enough). This would also prevent data moving between threads/cores.

It is likely possible to convert some of the current icache intensive code to data-intensive, when no complex calculations are needed (e.g. when dumping flops).

Other optimizations are also likely possible.

I would suggest hand-hacking improvements to see what they give before implementing for real.  E.g. in many cases where I have tried to reduce icache in the past the alternative performed worse.

@gezalore may have additional comments.

jcteng, 2024-08-17:
Thank you for the suggestion. In fact, using a 2-stage approach to aggregate fst fragments is also a method, which means forming fragment data in different threads first, and during the simulation process, there is no need to view the waveform, so there is no need to occupy the simulation process time to complete the aggregation. After the simulation is completed, these fragments can be aggregated through a second program, which can avoid the alternation of simulation and data encoding.

This may be a part of feasible measures that can enhance the parallelism in data retrieval.

But for the current situation we're facing, scattered capture data takes up a lot of time (mainly due to the inefficiency of the capture program's instructions), so the method mentioned above is likely not to directly address the issues I've pointed out above.

jcteng, 2024-08-17:
rtl project have been uploaded  [git@github.com:jcteng/rtls.git](https://github.com/jcteng/rtls.git)

closed 2025-09-10 by gezalore