verilator/verilator #5381: Improper continuous assignment bug

AndrewNolte, 2024-08-16:
Test case here: https://github.com/verilator/verilator/pull/5380
```
// this should be an initial assignment, but verilator thinks it's a continous assignment
logic temp0 = bar_in.field0;
// If it is observed (verilator public, coverage, etc.), then it switches correctly to initial
// logic temp0 /* verilator public */  = bar_in.field0;
```
It's almost never intended to assign initial with another signal, however it is technically the correct interpretation of the syntax. Slang has a [warning](https://sv-lang.com/warning-ref.html#static-init-value) for this.

Verilator interprets it as continuous, but interprets it correctly as initial when run under public, coverage, and perhaps other observability modifiers. 

I think the verilator test suite should have a few different testing environments to override some flags that shouldn't change the output of tests- like running with public, coverage, or tracing. It could also have a mode that randomly publicizes some wires and not others. We discovered this bug by running with an increased observability mode, causing some tests to fail.

wsnyder, 2024-08-18:
I suspect it's just random ordering decisions that make this work sometimes.

Thanks for the test case.  Might you look into a fix?

paul-demo, 2025-02-07:
This construct (initializing to some struct field which is a non-const value) is typically not synthesiable. I've even had discrepancies in commercial tools initializing to const values like `$bits(some_interface.some_signal)` so it wouldn't surprise me that initializing to `some_struct.some_field` is also not well tested.

Regardless, I think the bug is probably in this block of code:
https://github.com/verilator/verilator/blob/5b65c15ecaa0073f3185c7dcf08a16e216bfd44f/src/V3Const.cpp#L2837-L2860

closed 2026-04-27 by gezalore