verilator/verilator #5382: Accessing a two dimensional return value from a function leads to syntax error

goekce, 2024-08-17:
```sv
module m;                                                                                                                                                                                                                                    
  typedef byte o_t[1][1];                                                                                                                                                                                                                    
  function o_t f1(input byte i);                                                                                                                                                                                                             
    return '{'{1}};                                                                                                                                                                                                                          
  endfunction                                                                                                                                                                                                                                
  function o_t f2;                                                                                                                                                                                                                           
    return '{'{1}};                                                                                                                                                                                                                          
  endfunction                                                                                                                                                                                                                                
                                                                                                                                                                                                                                             
  initial $display(f1(0)[0][0]);                                                                                                                                                                                                             
//%Error: m.sv:10:28: syntax error, unexpected '[', expecting ')' or ','                                                                                                                                                                     
//   10 |   initial $display(f1(0)[0][0]);                                                                                                                                                                                                   
//      |                            ^                                                                                                                                                                                                       
                                                                                                                                                                                                                                             
  initial $display(f2[0][0]);                                                                                                                                                                                                                
//%Error: m.sv:12:20: Found definition of 'f2' as a FUNC but expected a variable                                                                                                                                                             
//   12 |   initial $display(f2[0][0]);                                                                                                                                                                                                      
//      |                    ^~                                                                                                                                                                                                              
//%Error: m.sv:12:20: Unsupported in C: Variable has the same name as function: 'f2'                                                                                                                                                         
//   12 |   initial $display(f2[0][0]);                                                                                                                                                                                                      
//      |                    ^~                                                                                                                                                                                                              
//        m.sv:6:16: ... Location of original declaration                                                                                                                                                                                    
//    6 |   function o_t f2;                                                                                                                                                                                                                 
//      |                ^~                                                                                                                                                                                                                  
                                                                                                                                                                                                                                             
endmodule
```
Linting above with `verilator --lint-only *.sv` leads to the above inline pasted errors.

Platform: Archlinux with `Verilator 5.027 devel rev v5.026-136-g48c71ef76`

Is this only related to the parser?

wsnyder, 2024-08-18:
The IEEE grammar has:

          function_subroutine_call [ '[' range_expression ']' ]

that is, only one range is allowed.  Does this work on other simulators?  If so, please consider a pull to support this, but also ask IEEE if this is a mistake in the grammar.

goekce, 2024-08-18:
Strange 😕. The grammar indeed does not allow more than a single index after a function call. I asked IEEE (:= Dave):

https://verificationacademy.com/forums/t/grammar-does-not-support-function-calls-with-two-dimensional-select-e-g-f-0-0-bug-or-feature/48235

wsnyder, 2024-09-08:
Closing - unless IEEE indicates mistake in grammar.

goekce, 2024-09-08:
Dear @wsnyder didn't you take a look to Dave's post?

> According to the author of that enhancement, it was an oversight. It should have been select instead of `range_expression. That is what most tools had already implemented.

Please reopen.

wsnyder, 2024-09-08:
Ok, might you be willing to make a pull to implement this?

goekce, 2024-09-09:
I would be willing to, but it will probably take a long time. If some one else is interested, please PR.

closed 2024-09-09 by wsnyder