verilator/verilator #5385: Ambiguous error if an external virtual class function is used instead of a local function in a `let` statement

goekce, 2024-08-18:
```sv
virtual class C #(                                                                                                                                                                                                                           
    int N = 1                                                                                                                                                                                                                                
);                                                                                                                                                                                                                                           
  typedef byte o_t[N][N];                                                                                                                                                                                                                    
  static function o_t mm(input byte i[N][N]);                                                                                                                                                                                                
    o_t o;                                                                                                                                                                                                                                   
    return o;                                                                                                                                                                                                                                
  endfunction                                                                                                                                                                                                                                
endclass                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                             
module m #(                                                                                                                                                                                                                                  
    int N = 1                                                                                                                                                                                                                                
) (                                                                                                                                                                                                                                          
    output byte o                                                                                                                                                                                                                            
);
  // Local definition of the function above                                                                                                                                                                                                                                          
  typedef byte o_t[N][N];                                                                                                                                                                                                                    
  function o_t mm(input byte i[N][N]);                                                                                                                                                                                                       
    o_t o;                                                                                                                                                                                                                                   
    return o;                                                                                                                                                                                                                                
  endfunction                                                                                                                                                                                                                                
                                                                                                                                                                                                                                             
  initial begin                                                                                                                                                                                                                              
    let arr = C#(1)::mm(                                                                                                                                                                                                                     
        '{'{1}}                                                                                                                                                                                                                              
    ); let arr2 = mm(                                                                                                                                                                                                                        
        '{'{1}}                                                                                                                                                                                                                              
    );                                                                                                                                                                                                                                       
    o = arr[0][0];  // Commenting out this line fixes the error                                                                                                                                                                            
    o = arr2[0][0];                                                                                                                                                                                                                          
    $finish;                                                                                                                                                                                                                                 
  end                                                                                                                                                                                                                                        
endmodule
```
Leads to:
```
%Error: Internal Error: m_bkp.sv:24:9: ../V3Slice.cpp:276: Array initialization should have been removed earlier                                                                                                                             
   24 |         '{'{1}}                                                                                                                                                                                                                      
      |         ^~
```
But commenting out the commented line above fixes the error. Why does the error happen only when the virtual function is used?

Questa accepts the code above, but VCS and Xcelium error out. I did not investigate in detail why these simulators error out.

Platform: Archlinux with `Verilator 5.027 devel rev v5.026-136-g48c71ef76`

wsnyder, 2024-08-27:
The code should give a better error.  Pull would be appreciated to improve the warning, otherwise will get to it at some point...

goekce, 2024-08-27:
I don't have time for a pull request, so PR from someone else is welcome.

ArturBieniek4, 2026-09-02:
Works now, fixed by [#6900](https://github.com/verilator/verilator/pull/6900).

closed 2026-09-02 by ArturBieniek4