verilator/verilator #5386: Support for wired OR and wired AND nets

Mijyuoon, 2024-08-19:
Is there a specific reason why `wor`/`trior` and `wand`/`triand` net types aren't supported? (checked on the latest master branch)

I have a project that uses them in several places and having to replace them with workarounds isn't great.

wsnyder, 2024-08-19:
wor etc are generally not used by recent synthesizable designs, and as somewhat proof of that, you are first to ask. If you need this please consider submitting a pull to fix it (probably mostly work in V3Tristate), as we are unlikely otherwise to get to it given competing priorities.

Mijyuoon, 2024-08-19:
Thanks for letting me know, I might actually have a go at it when I have time.

nookfoo, 2024-09-11:
We would also interested in the support of this, as we are using `wor`!

closed 2024-10-09 by wsnyder