verilator/verilator #5392: 5.028: Build fails: ../V3ParseGrammar.cpp:68:2: error: "Need lex token names"

yurivict, 2024-08-22:
flex is identidied by configure as:
```
checking for flex... /usr/local/bin/flex
/usr/local/bin/flex --version = flex 2.6.4
```

I can't find flex invocation line in the log.

[log](https://freebsd.org/~yuri/verilator-5.028-build-failure.log)

clang-18
FreeBSD 14.1

wsnyder, 2024-08-22:
verilog.y defines YYERROR_VERBOSE, and your log shows "bison ... -d ..." to enable debug, so there must be some FreeBSD oddity in how bison works you'll need to debug, as I don't use freeBSD myself.

wsnyder, 2024-08-22:
(The error is because YYERROR_VEBOSE is not 1).

yurivict, 2024-08-25:
This was due to the port patch of unclear origin.
It builds fine now.

Thank you for your help!

closed 2024-08-25 by yurivict