verilator/verilator #5400: Simulation fails with error "Active region did not converge"

kolappanc, 2024-08-23:
**Description of the issue** 
I have an SRAM design which is 8 rows deep and 8 bits wide. I am stitching up the 2 instances to obtain an SRAM that is 8 rows deep and 16 bits wide.  I have instantiated 2 instances of the same module (sram_top). When I try to build the simulation (from RTL to c++ and makefiles), it builds successfully. But when I try to run it (make V{top_module_name}) , it throws "Active region did not converge" error. **But when I instantiate just one instance of the SRAM module (sram_top), the simulation runs correctly without the active region did not converge error .**

Code snippet for the code I am working with,
The module "sram_top1" is described in another file which is included in this file (using a `include). I have also instantiated this module "top_sram_wrapper" in a stimulus file which provides it clk_i and rst.
 
module top_sram_wrapper (
input clk_i,
input rn,
output reg done_read 
);
integer i;

reg done_write = 1'b0;
reg done = 1'b0;
reg dvdd = 1'b1;
reg dvss = 1'b0;

reg cs = 1'b0;
reg we = 1'b0;
reg [3:0] addr_temp;
reg [2:0] addr;
reg [3:0] wmask;
wire [31:0] wdata;
reg [31:0] d_out;

wire [7:0] d_in_0, d_in_1, d_in_2, d_in_3; 
wire [7:0] d_out_0, d_out_1, d_out_2, d_out_3;
reg csb_col0, csb_col1, csb_col2, csb_col3;
reg [31:0] mem [0:511];

// Display the signals 
  always @ (posedge clk_i) begin
        $display("Time: %0t | clk_i: %b | rst_ni: %b | addr_a: %b | mem_value: %h | a_wmask: %b | !we: %b", $time, clk_i, rn, addr, wdata, ~wmask, !we);
  end

// Loading the BRAM with program memory
  initial begin
    $display("Loading ram.");
    $readmemh("hello_test.vmem", mem);
    for  (i=0; i<512; i++) $display("%h",mem[i]);
    $display("BRAM loaded successfully");
  end

// generating address for writing and reading
  always @(posedge clk_i or negedge rn) begin
       if (!rn) begin
         wmask = 4'b0000;
         addr = 3'b0;
         cs = 0;
         we = 0;
         addr_temp = 0;
       end
       else begin
        if (!done) begin
            if (addr_temp == 4'b1000) begin
                done = 1;
            end
            else begin
                addr_temp = addr_temp + 1'b1;
                addr = addr_temp-1;
            end
        end
       end
  end

// Checking the simulation conditions  
 always @(posedge clk_i or negedge rn) begin
   if (rn) begin
    if (done && !done_write) begin
        $display("Write over...");
        done_write = 1'b1;
        done = 1'b0;
        addr = 3'b0;
        addr_temp = 0;
    end
    else if (done && done_write) begin
        $display("Read over...");
        done_read = 1'b1;
        done = 1'b0;
        addr = 3'b0;
        addr_temp = 0;
    end
    else begin
       if (!done_write) begin
         wmask = 4'b1111;
         cs = 1;
         we = 1;
       end
       else begin 
        wmask = 4'b1111;
        cs = 1;
        we = 0;
       end              
    end
   end
  end

assign wdata = mem[addr];

assign d_out[7:0] = d_out_0;
assign d_out[15:8] = d_out_1;
assign d_out[23:16] = d_out_2;
assign d_out[31:24] = d_out_3;
assign csb_col0 = !wmask[0];
assign csb_col1 = !wmask[1];
assign csb_col2 = !wmask[2];
assign csb_col3 = !wmask[3];
assign d_in_0 = wdata [7:0];
assign d_in_1 = wdata [15:8];
assign d_in_2 = wdata [23:16];
assign d_in_3 = wdata [31:24];

sram_top1 I1 (
.addr(addr), 
.d_in(d_in_0), 
.d_out(d_out_0), 
.clk(clk_i), 
.csb(csb_col0), 
.dvdd(dvdd), 
.dvss(dvss), 
.rn(rn), 
.web(!we)
);
sram_top1 I2 (
.addr(addr), 
.d_in(d_in_1), 
.d_out(d_out_1), 
.clk(clk_i), 
.csb(csb_col1), 
.dvdd(dvdd), 
.dvss(dvss), 
.rn(rn), 
.web(!we)
);
endmodule

**The errors are shown below** 
-V{t1,1}         'act' region trigger index 1 is active: @([hybrid] arails_top.x_390.control_unit.pre_buf_chain.INV_5.a_y)
-V{t1,2}         'act' region trigger index 35 is active: @([changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__bl or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__br or [changed] arails_top.x_390.SRAM_Bank.Dummy_Replica_1.replica_bitcell_rbl_wl.drive_mode or [changed] arails_top.x_390.rbl_wl)
-V{t1,3}         'act' region trigger index 36 is active: @([changed] arails_top.x_390.Port.port_addr.Vcelloutdecoder_wl0__y1 or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__bl or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__br or [changed] arails_top.x_390.SRAM_Bank.Dummy_Replica_1.replica_bitcell_wl1.drive_mode)
-V{t1,4}         'act' region trigger index 37 is active: @([changed] arails_top.x_390.Port.port_addr.Vcelloutdecoder_wl0__y0 or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__bl or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_1__br or [changed] arails_top.x_390.SRAM_Bank.Dummy_Replica_1.replica_bitcell_wl0.drive_mode)
-V{t1,5}         'act' region trigger index 38 is active: @([changed] arails_top.x_390.Port.port_addr.Vcelloutdecoder_wl0__y6 or [changed] arails_top.x_390.SRAM_Bank.VcellinptDummy_Replica_0__br or [changed] arails_top.x_390.SRAM_Bank.Dummy_Replica_0_2.replica_cell_1.drive_mode or [changed] arails_top.x_390.Vcellinpcontrol_unit__rbl_bl)

%Error: dut_verilog/arails_top_module.v:1: Active region did not converge.

**The command used** 
verilator -cc --exe -O2 -x-assign 0 -Wall  -Wno-ENUMVALUE --trace --assert --coverage -Wno-fatal --timing --relative-includes --top-module arails_top -f input.vc dut_verilog/arails_top.v sim_main.cpp

The verilator version is "Verilator 5.026 2024-06-15 rev v5.026"

**What OS ?**
Cent OS 7 

It would be great if you could please assist me in coming up with a fix for this problem.

wsnyder, 2024-08-24:
The error message you posted is indicating what signals/blocks that Verilator thinks is changing.  These don't seem to correlate with what you posted exactly, so I can't offer exact advise, but look at the signal/block names in that message for some logic that does not converge - e.g. feeds back on itself.

kolappanc, 2024-08-24:
In our SRAM design we have more signals that Verilator thinks is changing since we have a loop in our design that we cannot remove. We do have delays in that said loop that should prevent it from not converging. However, the problem we are facing is that when we use two or more SRAM instances, verilator shows this error, but using a single SRAM instance does not give this error, which implies that signals are converging.
**This particular design works with Icarus Verilog.**

wsnyder, 2024-08-24:
You'll need to debug further, there's nothing indicated that helps to give you a direct suggestion to solve the issue.

kolappanc, 2024-08-29:
Thank you for your response. We will debug further to solve this issue. Please close this thread.

closed 2024-08-29 by wsnyder