verilator/verilator #5403: make error V3Randomize

fawqati-spherical, 2024-08-25:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
see below

What 'verilator' command line do we use to run your example?
> git clone https://github.com/verilator/verilator
>  cd verilator
> git checkout stable
> 

What 'verilator --version' are you using?  Did you try it with the git master version?
latest stable

What OS and distribution are you using?
Linux Rocky-8-latest-amd64-base 4.18.0-477.27.1.el8_8.x86_64

May we assist you in trying to fix this in Verilator yourself?
yes

../V3Randomize.cpp:1269:21: note:   no known conversion for argument 1 from ?<brace-enclosed initializer list>? to ?RandomizeMode?
In file included from ../V3PchAstNoMT.h:29:
../V3Randomize.cpp:1262:88: error: no matching function for call to ?RandomizeVisitor::wrapIfMode(<brace-enclosed initializer list>, AstVar*, AstNodeStmt*&)?
         return VN_AS(wrapIfMode({.asInt = varp->user1()}, getRandModeVar(classp), stmtp),
                                                                                        ^
../V3Ast.h:81:92: note: in definition of macro ?VN_AS?
 #define VN_AS(nodep, nodetypename) (AstNode::privateAs<Ast##nodetypename, decltype(nodep)>(nodep))
                                                                                            ^~~~~
../V3Randomize.cpp:1269:21: note: candidate: ?static AstNode* RandomizeVisitor::wrapIfMode(RandomizeMode, AstVar*, AstNode*)?
     static AstNode* wrapIfMode(const RandomizeMode mode, AstVar* modeVarp, AstNode* stmtp) {
                     ^~~~~~~~~~
../V3Randomize.cpp:1269:21: note:   no known conversion for argument 1 from ?<brace-enclosed initializer list>? to ?RandomizeMode?
../V3Randomize.cpp: In static member function ?static AstNode* RandomizeVisitor::wrapIfConstraintMode(AstClass*, AstConstraint*, AstNode*)?:
../V3Randomize.cpp:1267:91: error: no matching function for call to ?RandomizeVisitor::wrapIfMode(<brace-enclosed initializer list>, AstVar*, AstNode*&)?
         return wrapIfMode({.asInt = constrp->user1()}, getConstraintModeVar(classp), stmtp);
                                                                                           ^
../V3Randomize.cpp:1269:21: note: candidate: ?static AstNode* RandomizeVisitor::wrapIfMode(RandomizeMode, AstVar*, AstNode*)?
     static AstNode* wrapIfMode(const RandomizeMode mode, AstVar* modeVarp, AstNode* stmtp) {
                     ^~~~~~~~~~
../V3Randomize.cpp:1269:21: note:   no known conversion for argument 1 from ?<brace-enclosed initializer list>? to ?RandomizeMode?
make[2]: *** [../Makefile_obj:401: V3Randomize.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/tools/downloads/verilator/src/obj_dbg'
make[1]: *** [Makefile:85: ../bin/verilator_bin_dbg] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/tools/downloads/verilator/src/obj_opt'
make[1]: *** [Makefile:79: ../bin/verilator_bin] Error 2
make[1]: Leaving directory '/tools/downloads/verilator/src'
make: *** [Makefile:147: verilator_exe] Error 2

wsnyder, 2024-08-27:
As we don't see this error can you try to make a pull request to fix it?  Maybe:

```return wrapIfRandMode(VN_AS(m_modp, Class), varp, assignp);
```

Try making a temporary first

```
   AstClass* classp const = VN_AS(m_modp, Class);
   return wrapIfRandMode(classp, varp, assignp);
```

DenizzzGuzell, 2024-08-29:
Hi @kbieganski  @wsnyder,

After the patch I have runned `make clean` then `make` and `wrapIfConstraintMode` method return statement throws error. Below you can see the error message taken when building Verilator after patching `V3Randomize.cpp`  as given above:

```
VERILATOR_SOLVER='"z3 --in"' -include V3PchAstNoMT.h -c ../V3Randomize.cpp -o V3Randomize.o
../V3Randomize.cpp: In static member function ‘static AstNode* RandomizeVisitor::wrapIfConstraintMode(AstClass*, AstConstraint*, AstNode*)’:
../V3Randomize.cpp:1217:91: error: no matching function for call to ‘RandomizeVisitor::wrapIfMode(<brace-enclosed initializer list>, AstVar*, AstNode*&)’
         return wrapIfMode({.asInt = constrp->user1()}, getConstraintModeVar(classp), stmtp);
                                                                                           ^
../V3Randomize.cpp:1219:21: note: candidate: ‘static AstNode* RandomizeVisitor::wrapIfMode(RandomizeMode, AstVar*, AstNode*)’
     static AstNode* wrapIfMode(const RandomizeMode mode, AstVar* modeVarp, AstNode* stmtp) {
                     ^~~~~~~~~~
../V3Randomize.cpp:1219:21: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘RandomizeMode’
../Makefile_obj:398: recipe for target 'V3Randomize.o' failed
make[2]: *** [V3Randomize.o] Error 1
make[2]: Leaving directory '/home/dguzel/verilator/src/obj_dbg'
Makefile:84: recipe for target '../bin/verilator_bin_dbg' failed
make[1]: *** [../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/home/dguzel/verilator/src'
Makefile:145: recipe for target 'verilator_exe' failed
make: *** [verilator_exe] Error 2
```

ccache version is 3.4.1
gcc version is 8.4.0
OS Version : Ubuntu 18.04.5 LTS

kbieganski, 2024-08-29:
Ah, missed the other one.

#5417 should fix it, note we don't support such old compilers.

closed 2024-08-29 by wsnyder

fawqati-spherical, 2024-09-06:
Gents, thanks for your patience and all the input / help. It's sincerely appreciated !

At the time, i had to get Verilator up and running for my team so i decided to clone a previous version release and that compiled and installed correctly.

FYI, since then, i tried installing your latest 5.029 with the fix and that worked perfectly.

A BIG THANK YOU AGAIN FROM US ALL !

Best,
Faisal

PR #5417 (2024-08-29): Fix `V3Randomize` compile error on old GCC (#5403)
Fixes #5403 on GCC 8.

No functional change intended.