verilator/verilator #5404: `configure` checking features with different `std=` then compilation

ktbarrett, 2024-08-25:
In out CI, we are seeing Verilator's `./configure` step use `-std=gnu++11` as the C++ standard for doing feature checks ([see here](https://github.com/cocotb/cocotb/actions/runs/10543210110/job/29210898523?pr=3861#step:17:147)). Later, it uses `std=gnu++17` for the compilation ([see here](https://github.com/cocotb/cocotb/actions/runs/10543210110/job/29210898523?pr=3861#step:24:992)). But the features availability changes between the two language standards. The check for coroutine support works with `-std=gnu++11 -fcoroutines-ts`, but `std=gnu++17 -fcoroutine-ts` does *not*.

It seems that if the check `checking whether g++ -std=gnu++11 accepts -std=gnu++17` passes, the remaining feature checks should be using `-std=gnu++17`. That would cause the coroutine feature check to fail with `-fcoroutine-ts` and pass with `-std=gnu++20` as intended by [this](https://github.com/verilator/verilator/issues/5190#issuecomment-2245209134).

wsnyder, 2024-08-25:
It looks like you are calling ./configure with the compiler being set to "g++ -std=gnu++11", so that's contributing to the problem.

The line about testing gnu++17 is only about finding the most recent compiler.  This is intentionally not used in later steps as it breaks many standard libraries - generally using any std= will cause problems, as you've found ;)

ktbarrett, 2024-08-25:
We are not setting the compiler to that. It is being done by the configure script automatically ([see here](https://github.com/cocotb/cocotb/actions/runs/10543210110/job/29210898523?pr=3861#step:17:116)). This line reads `none needed` instead of `-std=gnu++11` when running on Linux.

wsnyder, 2024-08-26:
>checking for g++ option to enable C++11 features... -std=gnu++11

This is done by configure itself under the AC_PROG_CXX macro call Verilator uses. That it is needed suggests something is broken in the autoconf package, or the compiler on MacOS unrelated to Verilator.  Likely Apple (et al) added some hack to configure that is breaking things.

adachi6k, 2024-08-26:
I have faced similar problems when using clang++ on Linux.
When using coroutines in clang++, -std=gnu++20 should be specified at compilation time. But, because the options checked when verilated are either -fcoroutines-ts or -fcoroutines ([this](https://github.com/verilator/verilator/blob/d3e07b6ad9d6a2bbec65ddb783dee3059a9ec9b2/verilator-config.cmake.in#L350)), Unlike the check during verilator configure, the problem is that -fcoroutines is also specified in clang++.

At the moment, I applied the following patch, but it has not been tried in other cases and there may be a better way.

verilator/verilator-config.cmake.in

```cmake
if (${VERILATE_PREFIX}_TIMING)
    check_cxx_compiler_flag(-fcoroutines-ts COROUTINES_TS_FLAG)
    check_cxx_compiler_flag(-fcoroutines COROUTINES_FLAG)
    check_cxx_compiler_flag("-fcoroutines-ts -Wno-deprecated-experimental-coroutine" COROUTINES_EXPERIMENTAL_FLAG)
    check_cxx_compiler_flag(-std=gnu++20 GNU_STD20_FLAG)

    if (GNU_STD20_FLAG)
        target_compile_options(${TARGET} PRIVATE -std=gnu++20)
    elseif (COROUTINES_EXPERIMENTAL_FLAG)
        target_compile_options(${TARGET} PRIVATE "-fcoroutines-ts -Wno-deprecated-experimental-coroutine")
    elseif (COROUTINES_FLAG)
        target_compile_options(${TARGET} PRIVATE -fcoroutines)
    elseif (COROUTINES_TS_FLAG)
        target_compile_options(${TARGET} PRIVATE -fcoroutines-ts)
    endif()
endif()
```

wsnyder, 2024-08-27:
Thanks for the example - making some intermediate flag variables (COROUTINES_FLAGS etc) is a good idea.

For the final fix we'll need something that doesn't add any of them if coroutines compiles ok with no flags added.

thecaralice, 2024-09-04:
I'm not sure if this is the same problem; using verilator with clang++16 without providing any compiler-related options failes with "std::coroutine not found"-ish errors, although that is fixed by `-CFLAGS '-std=c++20'`, while using verilator with clang++18 fails with "unknown flag `-fcoroutines-ts`" which is not fixed by providing the same thing. However, I don't know a lot about Verilog or C++, so might be an issue on my side.

wsnyder, 2024-09-04:
@thecaralice does the above patch work?

thecaralice, 2024-09-05:
That is just a snippet of CMake code, not a proper patch.

thecaralice, 2024-09-05:
Assuming this is the actual patch, verilator behaves the same way after applying it, the compiler flags are the same as before.

```patch
--- a/verilator-config.cmake.in
+++ b/verilator-config.cmake.in
@@ -348,5 +348,17 @@
   if (${VERILATE_PREFIX}_TIMING)
     check_cxx_compiler_flag(-fcoroutines-ts COROUTINES_TS_FLAG)
+    check_cxx_compiler_flag(-fcoroutines COROUTINES_FLAG)
+    check_cxx_compiler_flag("-fcoroutines-ts -Wno-deprecated-experimental-coroutine" COROUTINES_EXPERIMENTAL_FLAG)
+    check_cxx_compiler_flag(-std=gnu++20 GNU_STD20_FLAG)
+
-    target_compile_options(${TARGET} PRIVATE $<IF:$<BOOL:${COROUTINES_TS_FLAG}>,-fcoroutines-ts,-fcoroutines>)
+    if (GNU_STD20_FLAG)
+        target_compile_options(${TARGET} PRIVATE -std=gnu++20)
+    elseif (COROUTINES_EXPERIMENTAL_FLAG)
+        target_compile_options(${TARGET} PRIVATE "-fcoroutines-ts -Wno-deprecated-experimental-coroutine")
+    elseif (COROUTINES_FLAG)
+        target_compile_options(${TARGET} PRIVATE -fcoroutines)
+    elseif (COROUTINES_TS_FLAG)
+        target_compile_options(${TARGET} PRIVATE -fcoroutines-ts)
+    endif()
   endif()
 endfunction()

```

ktbarrett, 2024-09-05:
So I checked the autoconf Changelog and 2.70 introduced checking for C++11, and that change was reported as causing issues and reverted 3 months ago. It should be a part of the next release of autoconf.

ktbarrett, 2024-09-05:
I was not able to get CMake to build Verilator on my Mac: `FLEX_INCLUDE_DIR-NOTFOUND`.

thecaralice, 2024-09-06:
I use Nix for building Verilator ([source][drv])

[drv]: https://github.com/NixOS/nixpkgs/blob/master/pkgs/applications/science/electronics/verilator/default.nix

thuvasooriya, 2024-10-24:
faced similar issues with verilator when using it with nix in macOS. I didn't want to install all the build tools with homebrew in my system so I used nix to build it from git. some observations.

1. building from nix develop works (currently using with VERILATOR_ROOT)
2. running the built verilator requires the same clang environment that was used during build.
3. After building verilator I exited the nix shell and ran .configure again. now I'm able to use the git verilator with my system clang with fcoroutines ( not sure whether it's a bad practice or not but it works :) )

is there a way to dynamically allocate std= and fcoroutines flags depending on checks in runtime or some option to reconfigure it for prebuilt binaries?

wsnyder, 2024-10-24:
>is there a way to dynamically allocate std= and fcoroutines flags depending on checks in runtime or some option to reconfigure it for prebuilt binaries?

All compile flags are overwritable with appropriate makefile flags, or using e.g. "-CFLAGS", however this is likely a big pain for usability which is best avoided.  What problem are you solving that makes you want to do this?

thuvasooriya, 2024-10-24:
> What problem are you solving that makes you want to do this?

I'm trying to simulate a verilog cpu design with `--trace` but prebuilt verilator binaries from nixpkgs returns unsupported argument error for `-fcoroutines-ts` which is necessary for trace afaik. ( I'm using nixpkgs with home-manager in my mac machine) 

> using e.g. "-CFLAGS"

I tried using -CFLAGS --std=c++20, but it was unresponsive and didn't change the results.

I guess even if the verilator binary is compiled in a different clang from the system one it should be able to work with CFLAGS overriding but it didn't work in my case and as you mentioned it's a big pain too.

I think further testing is required. I'll try to update my findings when i have time to test.

just a random thought - since the .configure command was able to configure the verilator build artifacts in the git directory after the verilator was actually built to make it work correctly with the system clang, is there any way to expose this configure as a sub command through verilator to configure prebuilt verilator binaries to the current clang available in the system.

wsnyder, 2024-10-24:
>prebuilt verilator binaries from nixpkgs returns unsupported argument error for -fcoroutines-ts

The best route is to build verilator yourself then.  Also, likely the version they have is old, so you can get the recent improvements.