verilator/verilator #5410: False positive property assertion

7FM, 2024-08-28:
Basically I wrote a simple TB with a golden sample and a DUT. To compare the results of those two modules, I added the following assertion:
```
    assert property (@(posedge clk) divs_out1 == divs_out2);
```
which fails during simulation. However, when I take a look at the simulation trace, I see no mismatch in these two values.
Also, when I replace the assertion with the following code:
```
    always_ff @(posedge clk) begin
        if (divs_out1 != divs_out2) begin
            $display("DIVS failure, different outputs: out1=0x%0h, out2=0x%0h", divs_out1, divs_out2);
            $fatal();
        end
    end
```
The simulation completes successfully without a `$fatal()` being triggered.
So I might hit some bug here?

> Can you attach an example that shows the issue?

I have attached an example project to reproduce the issue.
Simple run `make` followed by `./sim_build/bitext_test/Vsim_bitext_test -t my_trace.fst`.
[example.zip](https://github.com/user-attachments/files/16784503/example.zip)

What 'verilator' command line do we use to run your example?

> What 'verilator --version' are you using?

```
Verilator 5.028 2024-08-21 rev v5.028
```
> Did you try it with the git master version?

I haven't tried the latest master commit yet.

wsnyder, 2024-08-28:
When I try this with the verilator main it works fine. It looks like you are driving clocks and data changes on the same edge which can cause races, please instead try driving data changes on negedge.

7FM, 2024-09-02:
driving data on the negedge works, thx