verilator/verilator #5416: Internal Error: ../V3Randomize.cpp:925: Invalid reference?

solomatnikov, 2024-08-29:
```
%Error: Internal Error: dv/z_txn.sv:84:34: ../V3Randomize.cpp:925: Invalid reference?
   84 |   size         <= size_t'($clog2(CLINE_DATAW/8));
      |                                  ^~~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
package z_pkg;
   localparam int SIZEW=4;
   localparam int CLINE_DATAW=512;

  typedef logic [SIZEW-1:0]     size_t;
   typedef enum {
      TXN_UNSET,
      TXN_GET
   } txn_type_t;
endpackage

import z_pkg::*;

class z_txn;
   rand txn_type_t   req_txn_type;
   rand size_t       size;

   static function z_txn generate_txn(txn_type_t txn_type);
      z_txn txn = new;
      int success = txn.randomize() with {
         req_txn_type == txn_type;
	 size         <= size_t'($clog2(CLINE_DATAW/8));
      };
      assert(success != 0);
      txn.req_opcode = txn_opcode(txn_type);
      return txn;
   endfunction
endclass
```

What 'verilator' command line do we use to run your example?

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.028 2024-08-21 rev v5.028`

What OS and distribution are you using?

`Linux 5.14.0-427.24.1.el9_4.x86_64`

wsnyder, 2024-08-29:
For build error see #5403 and comment there if that fixes it.

wsnyder, 2024-08-29:
Sorry, ignore that posted to wrong bug.

wsnyder, 2024-08-29:
@kbieganski can you look please?

PR #5418 (2024-08-29): Fix capturing params in `randomize() with` (#5416)
Fixes #5416.
closed 2024-08-29 by kbieganski