verilator/verilator #5419: Issue with "Too many digits for 32 bit number" error?

evgenybabin, 2024-08-30:
How may we help - what is your question?

Dear Verilator Developers!

I'm getting the following error from the tool which seems to be related to some issue dealing with constants longer than 32bits and I wonder if I'm doing something wrong or it's a known issue. 
Here are the details:

**Verilator error:**

%Error: <my-sv-source>: Too many digits for 32 bit number: 'h6D5A1BA6540E36AE7384C94779710E89BAB5778362E9B302C3A2CF202B5615A9DD5E8EF2E2EF40444F7C23BBB76A508BF48BE900D8A33DAE8829FB3C643771A6
   65 |     localparam logic [BigKeyWidth-1:0] BigKey = BigKeyWidth'h6D5A1BA6540E36AE7384C94779710E89BAB5778362E9B302C3A2CF202B5615A9DD5E8EF2E2EF40444F7C23BBB76A508BF48BE900D8A33DAE8829FB3C643771A6;
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

**Relevant code snippet:** 

    localparam BigKeyWidth = 512;
    localparam logic [BigKeyWidth-1:0] BigKey = 'h6D5A1BA6540E36AE7384C94779710E89BAB5778362E9B302C3A2CF202B5615A9DD5E8EF2E2EF40444F7C23BBB76A508BF48BE900D8A33DAE8829FB3C643771A6;

**Current work around:** - adding the actual length of the number:

localparam logic [BigKeyWidth-1:0] BigKey = **512**'h6D5A1BA6540E36AE7384C94779710E89BAB5778362E9B302C3A2CF202B5615A9DD5E8EF2E2EF40444F7C23BBB76A508BF48BE900D8A33DAE8829FB3C643771A6;

Would appreciate your input on this.

Best regards,
Evgeny.

wsnyder, 2024-08-30:
This is not legal verilog, you need a size for numbers > 32 bits, e.g. "1024'h6....."

closed 2024-08-30 by wsnyder

evgenybabin, 2024-08-30:
Really appreciate such a prompt response!
Could you please point me to the official verilog spec stating this if you have it handy?
Thank you!
Evgeny.

wsnyder, 2024-08-30:
We have a better warning for decimal, I just fixed it for hex:

```
%Error: t/t_lint_unsized_bad.v:8:22: Too many digits for 32 bit number: '''h123456789123456789123456789'
        ... As that number was unsized ('...) it is limited to 32 bits (IEEE 1800-2023 5.7.1)
        ... Suggest adding a size to it.
```

evgenybabin, 2024-08-30:
This is great - thank you!
So about raising my previous question again: could you please point me to the official verilog spec stating this if you have it handy?
Thank you!
Evgeny.

wsnyder, 2024-08-30:
The spec section is in the error I posted: (IEEE 1800-2023 5.7.1)