verilator/verilator #5420: test_regress/../include/verilated_random.cpp:365: Internal: Solver error: (error "line 11 column 330: Argument size at position 1 has sort (_ BitVec 4) it does not match declaration (declare-fun bvshl ((_ BitVec 37) (_ BitVec 37)) (_ BitVec 37))")

solomatnikov, 2024-08-30:
```
driver: Leaving directory '.../verilator/test_regress/obj_vlt/t_constraint_operators'
	obj_vlt/t_constraint_operators/Vt_constraint_operators    > obj_vlt/t_constraint_operators/vlt_sim.log
%Warning: .../verilator/test_regress/../include/verilated_random.cpp:365: Internal: Solver error: (error "line 11 column 330: Argument size at position 1 has sort (_ BitVec 4) it does not match declaration (declare-fun bvshl ((_ BitVec 37) (_ BitVec 37)) (_ BitVec 37))")
*-* All Finished *-*
- t/t_constraint_operators.v:58: Verilog $finish
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2023 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

package z_pkg;
   localparam int ADDRW=37;
   localparam int SIZEW=4;

   typedef logic [ADDRW-1:0] address_t;
   typedef logic [SIZEW-1:0] size_t;

   typedef enum {
      TXN_0,
      TXN_1,
      TXN_2,
      TXN_3,
      TXN_4,
      TXN_5,
      TXN_6,
      TXN_7
   } txn_type_t;
endpackage

import z_pkg::*;

class Packet;
   rand txn_type_t   req_txn_type;
   rand size_t       size;
   rand address_t    address;

   constraint con_address {
      req_txn_type inside {TXN_1, TXN_2, TXN_3, TXN_4, TXN_5} -> address%(1<<size) == 0;
      req_txn_type inside {TXN_6, TXN_7} -> address%(1<<10) == 0;
   }

   static function Packet generate_txn(txn_type_t txn_type);
      Packet txn = new;
      int success = txn_type == TXN_0 ? txn.randomize() :
                                        txn.randomize() with { req_txn_type == txn_type;
                                                               size == 6; };
      assert(success != 0);
      return txn;
   endfunction // generate_txn
endclass

module t (/*AUTOARG*/);

   Packet p;

   int v;

   initial begin
      p = Packet::generate_txn(TXN_1);

      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

What 'verilator' command line do we use to run your example?

`test_regress/../bin/verilator --prefix Vt_constraint_operators ../obj_vlt/t_constraint_operators/Vt_constraint_operators__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_constraint_operators --fdedup --debug-check --comp-limit-members 10 --clk clk  -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_constraint_operators +define+TEST_DUMPFILE=obj_vlt/t_constraint_operators/simx.vcd t/t_constraint_operators.v`

What 'verilator --version' are you using?  Did you try it with the git master version?

`v5.029-gcdfb2221b`

What OS and distribution are you using?

`Linux 5.14.0-427.31.1.el9_4.x86_64`

solomatnikov, 2024-08-30:
If `soft` is added, then it errors out:

```
%Error: t/t_constraint_operators.v:34:66: syntax error, unexpected soft, expecting TYPE-IDENTIFIER
   34 |       req_txn_type inside {TXN_1, TXN_2, TXN_3, TXN_4, TXN_5} -> soft address%(1<<size) == 0;
      |                                                                  ^~~~
```

solomatnikov, 2024-08-30:
Both versions work with VCS on https://edaplayground.com

wsnyder, 2024-08-30:
@kbieganski 

As to `soft` I don't see an obvious reason that shouldn't parse, I can look at that if you'd like.

kozdra, 2024-09-06:
As for the `soft` syntax error, this is a limitation of our parsing, as there is a parser conflict between `expr: expr "->" expr` and `constraint_expr: expr "->" constraint_expr | ["soft"] expr`, which is avoided by omitting the problematic part of the latter rule. You can work around it by using an explicit `if` instead of an implication (`a -> soft b;`  ---> `if (a) soft b;`). Note that soft constraints are not yet supported in Verilator.

And for the bit vector width problem, I am taking a look at it.

solomatnikov, 2024-09-20:
> As for the `soft` syntax error, this is a limitation of our parsing, as there is a parser conflict between `expr: expr "->" expr` and `constraint_expr: expr "->" constraint_expr | ["soft"] expr`, [...]

Is there any plan to support `a -> soft b;` in verilator?

closed 2026-03-16 by wsnyder