verilator/verilator #5429: Extreme performance loss when upgrading to gcc-13

joern274, 2024-09-03:
When upgrading from gcc-11 to gcc-13 (with other words : from Ubuntu 22.04 to Ubuntu 24.04) we noticed that the time to compile code generated from Verilator increased by a factor of five. The ftime report (see attached) tells us that most of the time is spent for "reload CSE regs" which was insignificant when running good old gcc-11. I must admit I don't know what this means.

Is there anything which can be done on the Verilator side to avoid this extreme performance loss (other than downgrading the compiler) ?

I have tested several Verilator versions including the latest master. While compilation time between Verilator versions varies only slightly there is always this significant (factor five) drop when migrating from gcc-11 to gcc-13.
[project_analytic_filter.tar.gz](https://github.com/user-attachments/files/16851369/project_analytic_filter.tar.gz)

wsnyder, 2024-09-03:
I don't think I've seen that issue.  Can you please file a bug at gcc.gny.org/bugzilla and post them the appropriate code (if public)?

Usually they can suggest a flag to disable that optimization, or alternatively how Verilator can emit code around it.

wsnyder, 2024-09-03:
http://gcc.gnu.org/bugzilla

joern274, 2024-09-03:
Thanks for quick response. Sure, the attached code can be posted at gcc site as you suggested.

The performance loss I reported is also visible when removing the optimization flag.

wsnyder, 2024-09-03:
Just to be sure you mean removing the GCC's optimization flag running the compile command standalone.  Note Verilator optimization level is different, and the Verilator makeflags usually have some optimization defaulted on.

joern274, 2024-09-03:
Yes, what I tested was whether removing GCC's -O3 flag makes a difference (it does not). As you can see in the attached file I isolated the g++ command from the Verilator-generated Makefile which took most time and played with the compiler flags. However - I have no clue how the compiler works internally and which buttons I should push.

Have asked for a GCC user account to report the problem on their site.

wsnyder, 2025-08-21:
I'm presuming this was reported upstream.

closed 2025-08-21 by wsnyder