verilator/verilator #5432: Support `disabling fork by name`

WisniewskiP, 2024-09-05:
In release v5.018 you added support for `disable fork`, would it be possible to extend it to support `disabling fork by name`?

I would like to use VUnit but it fails with message:
```txt
%Error-UNSUPPORTED: vunit_pkg.sv:222:13: Unsupported: disabling fork by name
  222 |             disable wait_or_timeout;
```
I tried it with Verilator v5.028.

wsnyder, 2024-09-05:
@kozdra?  

@WisniewskiP if he doesn't have that in flight, perhaps you'd consider a pull request to implement it, or at least a pull to add tests for the feature to prove the implementation?

kozdra, 2024-09-06:
I do not have it planned for the moment. @WisniewskiP if you want to start some work on it, you can refer to IEEE 1800-2023 9.6.2, and it might be worth starting with the descendant processes that are already tracked for the purposes of disable fork. It should be possible to use the same process tracking infrastructure for the implementation.

WisniewskiP, 2024-09-13:
I have looked at the the PR for disable to get some feeling what need to be done. I wasn't able to find the right spot to extend it. I need to spend some more time to analyze how process tracking is implemented.
Unfortunately at this moment I don't have time to do it.