verilator/verilator #5433: Configuration Files with `-file` Must Be Included First

sifferman, 2024-09-05:
```
Ubuntu 22.04.4 LTS
Verilator 5.027 devel rev v5.026-4-g833df6d9e
```
Hello. It seems that `lint_off -file` doesn't work on source files that have already been passed as argument.

## Example:

```systemverilog
// bad.sv
module bad(input a);
endmodule
```
```systemverilog
// lint.vlt
`verilator_config
lint_off -rule UNUSEDSIGNAL -file "bad.sv"
```
```
// verilator.f (lint passes)
lint.vlt // first
bad.sv
--lint-only
-Wall
```
```
// verilator.f (lint fails)
bad.sv
lint.vlt // second
--lint-only
-Wall
```
```log
verilator -f verilator.f
%Warning-UNUSEDSIGNAL: bad.sv:2:18: Signal is not used: 'a'
                                : ... note: In instance 'bad'
    2 | module bad(input a);
      |                  ^
                     ... For error description see https://verilator.org/warn/UNUSEDSIGNAL?v=5.027
```

I assume this is undesired behavior. I won't have time to look into this until October

sifferman, 2024-09-05:
Actually, just noticed this may be a duplicate of https://github.com/verilator/verilator/issues/5185

wsnyder, 2024-09-06:
Agree, duplicates #5185.

closed 2024-09-06 by wsnyder