verilator/verilator #5435: Associative Array Next/Prev Bugs

sifferman, 2024-09-06:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.028-49-g1c38744fd
```

Hello. I have been playing around with the associative array .next and .prev methods and have noticed a couple bugs. I figured these bugs were similar enough to lump into 1 issue.

## Working Example

```systemverilog
module t;

    typedef bit [7:0] data_t;
    data_t dict [data_t];

    initial begin
        data_t a;

        dict.first(a);
        dict.next(a);

        $finish;
    end

endmodule
```

## Argument with initial value causes g++ error

```systemverilog
module t;

    typedef bit [7:0] data_t;
    data_t dict [data_t];

    initial begin
        data_t a;

        a = 4; // set to number instead of .first
        dict.next(a);

        $finish;
    end

endmodule
```
```
g++ -Os  -I.  -MMD -Iverilator/include -Iverilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o Vnext_casting__ALL.o Vnext_casting__ALL.cpp
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp: In function ‘void Vnext_casting___024root___eval_initial__TOP(Vnext_casting___024root*)’:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:35:70: error: cannot bind non-const lvalue reference of type ‘unsigned char&’ to a value of type ‘unsigned int’
   35 |     t__DOT__unnamedblk1__DOT__a_exists = vlSelfRef.t__DOT__dict.next(4U);
      |                                                                      ^~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:978:21: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::next(T_Key&) const [with T_Key = unsigned char; T_Value = unsigned char]’
  978 |     int next(T_Key& indexr) const {
      |              ~~~~~~~^~~~~~
```

## Implicit cast causes g++ error

```systemverilog
module t;

    typedef bit [7:0] data_t;
    data_t dict [data_t];

    initial begin
        int a; // int instead of data_t

        dict.first(a);
        dict.next(a); // implicit cast from int to data_t

        $finish;
    end

endmodule
```
```
g++ -Os  -I.  -MMD -Iverilator/include -Iverilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o Vnext_casting__ALL.o Vnext_casting__ALL.cpp
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp: In function ‘void Vnext_casting___024root___eval_initial__TOP(Vnext_casting___024root*)’:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:35:41: error: cannot bind non-const lvalue reference of type ‘unsigned char&’ to a value of type ‘unsigned int’
   35 |     vlSelfRef.t__DOT__dict.first((0xffU & vlSelfRef.t__DOT__unnamedblk1__DOT__a));
      |                                  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:964:22: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::first(T_Key&) const [with T_Key = unsigned char; T_Value = unsigned char]’
  964 |     int first(T_Key& indexr) const {
      |               ~~~~~~~^~~~~~
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:38:71: error: cannot bind non-const lvalue reference of type ‘unsigned char&’ to a value of type ‘unsigned int’
   37 |                                                                      (0xffU
      |                                                                      ~~~~~~
   38 |                                                                       & vlSelfRef.t__DOT__unnamedblk1__DOT__a));
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:978:21: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::next(T_Key&) const [with T_Key = unsigned char; T_Value = unsigned char]’
  978 |     int next(T_Key& indexr) const {
      |              ~~~~~~~^~~~~~
```

wsnyder, 2024-09-06:
The first error is due to a misoptimization in propagating the constant.  Perhaps you'd be willing to make a pull to fix this?

The second should result in a compilation error as the types are not compatible - at least one other simulator correctly reports this.  Can you please file a separate bug?

closed 2024-09-07 by wsnyder

sifferman, 2024-09-07:
Thanks for the help!