verilator/verilator #5437: Support for Constrained Randomization of Unpacked Arrays

YilouWang, 2024-09-06:
Hello,

Since constrained random for packed arrays is now supported with the addition of astsel emitSMT(), I am now starting to explore the issue of constrained random for unpacked arrays. 

This is a relatively complex problem because when array becomes unpacked, the best approach is not to treat it as a large chunk of data, but rather as an array data type. This approach will also help in solving issues related to dynamic arrays in the future.

The smt-lib2 logic QF_ABV supports arrays, allowing array reads and writes through select and store.

So far, I have modified ConstraintVisitor and parts of verilated_random.cpp, and I have successfully sent the correctly formatted array constraints to the solver and received correct results.

The current issue is that when parsing the solution, I need to assign each value of the unpacked array to the corresponding verilated variables.

Does anyone have any ideas on how this step should be done?

Current progress:
Input to solver:
```
(set-option :produce-models true)
(set-logic QF_ABV)
(define-fun __Vbv ((b Bool)) (_ BitVec 1) (ite b #b1 #b0))
(define-fun __Vbool ((v (_ BitVec 1))) Bool (= #b1 v))
(declare-fun unpacked_array () (Array (_ BitVec 32) (_ BitVec 32)))
(assert (= #b1 (__Vbv (= (select unpacked_array #x00000000) #x00000007))))
(assert (= #b1 (__Vbv (= (select unpacked_array #x00000001) #x00000008))))
(check-sat)
```
Output from solver:
```
((unpacked_array (store ((as const (Array (_ BitVec 32) (_ BitVec 32))) #x00000007)
       #x00000001
       #x00000008)))
```

wsnyder, 2024-09-06:
Good progress!

An example of setting indices is verilated.cpp's VL_READMEM_N. Also can look at verilated_vpi.cpp specifically vpi_handle_by_index which returns a handle to an array index and the created handle's code.

CC @kozdra as to parsing randomization specifically.

kozdra, 2024-09-09:
@YilouWang Have you looked at #5413? It takes a similar approach, however avoids parsing complex SMT structures by using selects in `get-values`. I think when that is merged, packed and unpacked arrays could work very similar to queues.

wsnyder, 2024-10-02:
Thanks for working on this!  Might you consider doing the queue and dynamic arrays next (likely other than testing, same code will handle both) (#4625)?

closed 2024-10-02 by wsnyder

YilouWang, 2024-10-02:
Yes, I would be happy to. In fact, I’m already working on this, and I will also open an issue about support for queues. Thanks!