verilator/verilator #5438: Support `std::randomize` outside of class

steven-bellock, 2024-09-06:
>What would you like added/supported?

The ability to call `std::randomize` outside of a class.
>What 'verilator --version' are you using?  

Verilator 5.020 2024-01-01 rev (Debian 5.020-1)
>Did you try it with the git master version?

No.

>Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
```
module testbench;
    initial begin
        logic a;
        std::randomize(a);
        $finish;
    end
endmodule : testbench
```
produces
```
%Error: tb.sv:4:14: Calling implicit class method 'randomize' without being under class
    4 |         std::randomize(a);
```
in Verilator.
>May we assist you in trying to fix this in Verilator yourself?

Sure. Not opposed to it.

wsnyder, 2024-09-06:
Great! @kozdra might you be able to advise as to how @steven-bellock can get started?

kozdra, 2024-09-09:
Unfortunately std::randomize is not yet supported (one previous feature request being #4706), and the current error message on master (#5354) is:
```
%Warning-CONSTRAINTIGN: issue5438.sv:4:14: std::randomize ignored (unsupported)
                                         : ... note: In instance 'testbench'
    4 |         std::randomize(a);
      |              ^~~~~~~~~
                        ... For warning description see https://verilator.org/warn/CONSTRAINTIGN?v=5.029
                        ... Use "/* verilator lint_off CONSTRAINTIGN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

As a workaround for current Verilator, you can create a wrapper class to hold your random variables.

However, if you wish to implement the missing feature, you can start by taking inspirations from #4999 (somewhat outdated). The simplest case would be randomizing a single integer variable, and once that works, adding more cases (enums, arrays, etc., currently done for classes by V3Randomize.cpp - ideally the same code could be used by both randomize syntaxes eventually).
The areas of Verilator that would need attention are mostly V3Width::visit(AstNodeFTaskRef*) (where the error message is currently printed) and V3Randomize.

wsnyder, 2025-04-15:
@YilouWang perhaps you'd be interested in adding this (& see also #4706).

YilouWang, 2025-04-16:
Thank you for the mention! Yes, we’d be happy to help with this. 😊

closed 2025-07-25 by wsnyder

steven-bellock, 2025-07-25:
Thanks @YilouWang !