verilator/verilator #5439: Fix associative array methods not erroring on incompatible types

sifferman, 2024-09-06:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.028-49-g1c38744fd
```

Associative array methods do not currently report incompatible types. Though the LRM does not make it very clear what is incompatible. It simply states that the input should be "an index of the appropriate type for the array in question."

I would expect that an implicit cast is allowed only if
* if the input data type is integral, either 2-state or 4-state
* and its width is at least the width of the array key type

Otherwise there should be an error reported.

## Example

```systemverilog
module t;

    typedef bit [31:0] data_t;
    data_t dict [data_t];

    // verilator lint_off WIDTHTRUNC
    // verilator lint_off WIDTHEXPAND
    initial begin
        bit [31:0] a;
        logic [63:0] b;
        bit [7:0] c; // int instead of data_t

        dict.first(a); // correctly does not produce error, but doesn't compile
        dict.next(a); // correctly does not produce error, but doesn't compile

        dict.first(b); // correctly does not produce error, but doesn't compile
        dict.next(b); // correctly does not produce error, but doesn't compile

        dict.first(c); // incorrectly does not produce error
        dict.next(c); // incorrectly does not produce error

        $finish;
    end

endmodule
```
```
g++ -Os  -I.  -MMD -Iverilator/include -Iverilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o Vnext_casting__ALL.o Vnext_casting__ALL.cpp
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp: In function ‘void Vnext_casting___024root___eval_initial__TOP(Vnext_casting___024root*)’:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:34:34: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to an rvalue of type ‘IData’ {aka ‘unsigned int’}
   34 |     vlSelfRef.t__DOT__dict.first((IData)(vlSelfRef.t__DOT__unnamedblk1__DOT__b));
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:964:22: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::first(T_Key&) const [with T_Key = unsigned int; T_Value = unsigned int]’
  964 |     int first(T_Key& indexr) const {
      |               ~~~~~~~^~~~~~
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:35:33: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to an rvalue of type ‘IData’ {aka ‘unsigned int’}
   35 |     vlSelfRef.t__DOT__dict.next((IData)(vlSelfRef.t__DOT__unnamedblk1__DOT__b));
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:978:21: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::next(T_Key&) const [with T_Key = unsigned int; T_Value = unsigned int]’
  978 |     int next(T_Key& indexr) const {
      |              ~~~~~~~^~~~~~
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:36:44: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to a value of type ‘CData’ {aka ‘unsigned char’}
   36 |     vlSelfRef.t__DOT__dict.first(vlSelfRef.t__DOT__unnamedblk1__DOT__c);
      |                                  ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:964:22: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::first(T_Key&) const [with T_Key = unsigned int; T_Value = unsigned int]’
  964 |     int first(T_Key& indexr) const {
      |               ~~~~~~~^~~~~~
In file included from Vnext_casting__ALL.cpp:8:
Vnext_casting___024root__DepSet_hd9543883__0__Slow.cpp:37:43: error: cannot bind non-const lvalue reference of type ‘unsigned int&’ to a value of type ‘CData’ {aka ‘unsigned char’}
   37 |     vlSelfRef.t__DOT__dict.next(vlSelfRef.t__DOT__unnamedblk1__DOT__c);
      |                                 ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from verilator/include/verilated.h:1018,
                 from Vnext_casting__pch.h:21,
                 from Vnext_casting.cpp:4,
                 from Vnext_casting__ALL.cpp:3:
verilator/include/verilated_types.h:978:21: note:   initializing argument 1 of ‘int VlAssocArray<T_Key, T_Value>::next(T_Key&) const [with T_Key = unsigned int; T_Value = unsigned int]’
  978 |     int next(T_Key& indexr) const {
      |              ~~~~~~~^~~~~~
```
I don't have access to commercial tools at the moment to confirm this behavior.

paul-demo, 2025-10-23:
Cadence and Synopsys tools complain about incompatible types for `b` and `c` cases, and they work for the `a` case, as you point out. I would think that Verilator should be ok with any of them because you have waived WIDTHEXPAND (which applies to `c`) and WIDTHTRUNC (which applies to `b`), but that's based on my instinct and not based on any Verilog standard.

To match the commercial simulators, `a` should work; `b` and `c` should complain about invalid type conversion.

(Questa is fine with all three of these...)

wsnyder, 2026-02-02:
Per #6987, when fixing, also remember to type check default: assignments.

closed 2026-11-01 by wsnyder