verilator/verilator #5440: Is the problem is in this verilog file or in my verilator command line

zapta, 2024-09-06:
I am trying to lint a project that includes the yosys ice40 verilog library file https://github.com/YosysHQ/yosys/blob/main/techlibs/ice40/cells_sim.v and am getting lint errors. Is the problem in the file itself or is it just my verilator command line that is incorrect?  (e.g. selecting a different verilog standard)

```
$ verilator --lint-only   cells_sim.v
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1497:21: syntax error, unexpected '=', expecting ','
 1497 |  input  [15:0] MASK = 16'h 0000,
      |                     ^
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1533:2: syntax error, unexpected generate
 1533 |  generate
      |  ^~~~~~~~
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1607:2: syntax error, unexpected initial
 1607 |  initial begin
      |  ^~~~~~~
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1736:21: syntax error, unexpected '=', expecting ','
 1736 |  input  [15:0] MASK = 16'h 0000,
      |                     ^
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1761:2: syntax error, unexpected IDENTIFIER
 1761 |  SB_RAM40_4K #(
      |  ^~~~~~~~~~~
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1872:21: syntax error, unexpected '=', expecting ','
 1872 |  input  [15:0] MASK = 16'h 0000,
      |                     ^
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:1897:2: syntax error, unexpected IDENTIFIER
 1897 |  SB_RAM40_4K #(
      |  ^~~~~~~~~~~
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:2008:21: syntax error, unexpected '=', expecting ','
 2008 |  input  [15:0] MASK = 16'h 0000,
      |                     ^
%Error: /Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v:2033:2: syntax error, unexpected IDENTIFIER
 2033 |  SB_RAM40_4K #(
      |  ^~~~~~~~~~~
%Error: Exiting due to 9 error(s)
```

zapta, 2024-09-06:
Answering my own question. The flag -DNO_ICE40_DEFAULT_ASSIGNMENTS was missing in the command line. With it, the errors disappear and I can understand and know how to handle the new warnings.

closed 2024-09-06 by zapta

wsnyder, 2024-09-06:
If you try master this should already be fixed, see #5358