verilator/verilator #5446: Incorrect warning "Unsupported: randomizing this expression, treating as state"

solomatnikov, 2024-09-11:
```
./test_regress/t/t_randomize_method_with.py 2>&1 | tee log
==SUMMARY: Left 1  Passed 0  Failed 0  Time 0:00
driver.py: Found 24 cores, using -j 24
======================================================================
vlt/t_randomize_method_with: ==================================================
	export GCOV_PREFIX_STRIP='99'
	export GCOV_PREFIX='/work/sols/work/verilator/test_regress/obj_dist/gcov_1'
	perl /work/sols/work/verilator/test_regress/../bin/verilator --prefix Vt_randomize_method_with -cc -Mdir obj_vlt/t_randomize_method_with --fdedup --debug-check --comp-limit-members 10 --x-assign unique --make gmake --exe ../obj_vlt/t_randomize_method_with/Vt_randomize_method_with__main.cpp --fno-inline --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_randomize_method_with +define+TEST_DUMPFILE=obj_vlt/t_randomize_method_with/simx.vcd t/t_randomize_method_with.v   > obj_vlt/t_randomize_method_with/vlt_compile.log
%Warning-CONSTRAINTIGN: t/t_randomize_method_with.v:137:34: Unsupported: randomizing this expression, treating as state
                                                          : ... note: In instance 'mwith'
  137 |     c = cls.randomize() with { b inside {IDS}; };
      |                                  ^~~~~~
                        ... For warning description see https://verilator.org/warn/CONSTRAINTIGN?v=5.029
                        ... Use "/* verilator lint_off CONSTRAINTIGN */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
git diff
diff --git a/test_regress/t/t_randomize_method_with.v b/test_regress/t/t_randomize_method_with.v
index 616864d7b..d7612afa2 100644
--- a/test_regress/t/t_randomize_method_with.v
+++ b/test_regress/t/t_randomize_method_with.v
@@ -18,6 +18,10 @@ begin \
    if (ok != 1) $stop; \
 end
 
+//package cfg_pkg;
+   parameter int IDS[3] = {2, 3, 5};
+//endpackage
+
 class Boo;
   function new();
     boo = 6;
@@ -130,6 +134,8 @@ module mwith();
         $display("Failed to randomize foo with inline constraints");
     end
 
+    c = cls.randomize() with { b inside {IDS}; };
+
     if (cls.randomize() with { b == 1;} != 1) $stop;
```

What 'verilator' command line do we use to run your example?

`verilator --prefix Vt_randomize_method_with -cc -Mdir obj_vlt/t_randomize_method_with --fdedup --debug-check --comp-limit-members 10 --x-assign unique --make gmake --exe ../obj_vlt/t_randomize_method_with/Vt_randomize_method_with__main.cpp --fno-inline --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_randomize_method_with +define+TEST_DUMPFILE=obj_vlt/t_randomize_method_with/simx.vcd t/t_randomize_method_with.v`

What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.029 devel rev v5.028-72-g521c5c2f7`

What OS and distribution are you using?

`Linux 5.14.0-427.33.1.el9_4.x86_64`

solomatnikov, 2024-09-11:
Note: this compiles w/o problem:

```
git diff
diff --git a/test_regress/t/t_randomize_method_with.v b/test_regress/t/t_randomize_method_with.v
index 616864d7b..c511bf7cf 100644
--- a/test_regress/t/t_randomize_method_with.v
+++ b/test_regress/t/t_randomize_method_with.v
@@ -18,6 +18,10 @@ begin \
    if (ok != 1) $stop; \
 end
 
+//package cfg_pkg;
+   parameter int IDS[3] = {2, 3, 5};
+//endpackage
+
 class Boo;
   function new();
     boo = 6;
@@ -130,6 +134,8 @@ module mwith();
         $display("Failed to randomize foo with inline constraints");
     end
 
+    c = cls.randomize() with { b inside {2, 3, 5}; };
+
     if (cls.randomize() with { b == 1;} != 1) $stop;
     if (cls.b != 1) $stop;
     `check_rand(cls2, cls2.a);
```

solomatnikov, 2024-09-11:
In some cases I get an internal error on very similar code:

```
Internal Error: ... ../V3Randomize.cpp:925: Invalid reference?
  103 |                                                                    id inside {MMIO_ID}; };
      |                                                                               ^~~~~~~~

```

wsnyder, 2024-09-11:
@kozdra - perhaps this needs to be an unsupported as the parameter is basically a packed array?

kozdra, 2024-09-11:
@solomatnikov I happen to have fixed this in a patch I just submitted in #5448. For the other error message, I also have a WIP patch that I have been working on that might be related to your problem. I think I will submit it tomorrow.

kozdra, 2024-09-12:
@solomatnikov Unfortunately I do not have the capacity to submit this patch today, because it turned out to be more hacky than I thought. Although maybe the following patch can solve the second problem for you.

```patch
diff --git a/src/V3Randomize.cpp b/src/V3Randomize.cpp
index 5315b3b2f..5acd14db6 100644
--- a/src/V3Randomize.cpp
+++ b/src/V3Randomize.cpp
@@ -923,7 +928,7 @@ public:
         : m_argsp(nullptr)
         , m_callerp(callerp)
         , m_classp(classp)
-        , m_lookup(classp) {
+        , m_lookup(VN_IS(callerp, Class) ? VN_AS(callerp, Class) : classp) {
         iterateAndNextNull(nodep);
     }
 
```

solomatnikov, 2024-09-12:
@kozdra no worries, anyway I am working on other issues on my side.

PR #5448 (2024-09-19): Support inside array constraints (#5446)
This depends on #5431, but can as well be made part of that.

Closes #5446
closed 2024-09-19 by wsnyder