verilator/verilator #5456: Using repeat in fork-join_any causes a compile error.

katsui97, 2024-09-18:
A compile error occurs when a repeat statement is inside a fork-join_any. By the way, if it's a for statement, it passes. Also, if any is not used, the repeat statement passes as well.

Rev:  Verilator 5.029 devel rev v5.028-91-g7081ea0ed

-----------------Compilation result----------------------
%Error-LIFETIME: test.sv:19:9: Invalid reference: Process might outlive variable `__Vrepeat0`.
                             : ... note: In instance 'top'
                             : ... Suggest use it as read-only to initialize a local copy at the beginning of the process, or declare it as static. It is also possible to refer by reference to objects and their members.
   19 |         repeat(10)                     
      |         ^~~~~~
                 ... For error description see https://verilator.org/warn/LIFETIME?v=5.029
%Error: Exiting due to 1 error(s)

-------------- test bench ---------------
`timescale 1ns/100ps

module top();

  bit CLK;
  // Gen Clock
  always #10
    CLK = ~CLK;

  initial
  begin
    fork
      begin
        forever
          @(posedge CLK);
      end
      begin
        //for(int i=0; i < 10; ++i)
        repeat(10)
          @(posedge CLK);
      end
    //join
    join_any

    $finish;
  end

endmodule

wsnyder, 2024-09-19:
For historical reasons (predating begin statement variables), repeat made a variable underneath the module/class, instead of making a local variable.  Fixed.

closed 2024-09-19 by wsnyder

katsui97, 2024-09-19:
Oh, thank you for the prompt response. I have confirmed the effectiveness of the fix.

Rev: Verilator 5.029 devel rev v5.028-94-g371a4055b