verilator/verilator #5457: how to use with --timing

zyp2005, 2024-09-18:
The version in my computer is 5.008. When I learn systemverilog with delay, I try to simulate it. After searching the doc, I began to simulate. However, it appears that there might be an infinite loop occurring in the simulation.  What can I do? Is the code I written is error?

Here are files:
```verilog
`timescale 1ns/100ps

  module example(input logic a, b, c,
                 output logic y);
   logic                      ab, bb, cb, n1, n2, n3;

   assign #1 {ab, bb, cb} = ~{a, b, c};
   assign #2 n1 = ab & bb & cb;
   assign #2 n2 = a & bb & cb;
   assign #2 n3 = a & bb & c;
   assign #4 y = n1 | n2 | n3;

endmodule; // example
```

```c++
#include <stdio.h>
#include <verilated.h>
#include <verilated_fst_c.h>

#include "Vexample.h"

#define MAX_SIM_TIME 100

vluint64_t sim_time    = 0;
vluint64_t posedge_cnt = 0; 

void dut_reset(Vexample * dut, vluint64_t & sim_time); 

int main(int argc, char ** argv)
{
    VerilatedContext * contextp = new VerilatedContext;
    contextp->debug(0);
    contextp->commandArgs(argc,
                          argv);
    contextp->traceEverOn(true); 
    Verilated::mkdir("logs");    

    VerilatedFstC * m_trace = new VerilatedFstC; 

    Vexample * dut = new Vexample; 

    dut->trace(m_trace, 5); 
    m_trace->open("./logs/waveform.fst"); 

    for (int a = 0; a <= 1; a++)
    {
        for (int b = 0; b <= 1; b++)
        {
            for (int c = 0; c <= 1; c++)
            {
                printf("%d %d %d\n", a, b, c);
                dut->a = a;
                dut->b = b;
                dut->c = c;
                dut->eval();
                printf("before while\n");
                unsigned long long d = 0;
                while (dut->eventsPending())
                {
                    printf("%lu\n", dut->nextTimeSlot());
                    contextp->time(dut->nextTimeSlot());
                    dut->eval();
                }
                d = 0;
                printf("after while\n");
                m_trace->dump(contextp->time()); 
            }
        }
    }

    m_trace->close(); 
    delete dut;
    delete contextp;

    return 0;
}

```

Makefile
```Makefile
MODULE=example

.PHONY:sim
sim: ./logs/waveform.fst

.PHONY:verilate
verilate: .stamp.verilate

.PHONY:build
build: obj_dir/V$(MODULE)

.PHONY:waves
waves: ./logs/waveform.fst
	@echo
	@echo "### WAVES ###"
	gtkwave ./logs/waveform.fst

./logs/waveform.fst: ./obj_dir/V$(MODULE)
	@echo
	@echo "### SIMULATING ###"
	@./obj_dir/V$(MODULE)

./obj_dir/V$(MODULE): .stamp.verilate
	@echo
	@echo "### BUILDING SIM ###"
	make -C obj_dir -f V$(MODULE).mk V$(MODULE)

.stamp.verilate: $(MODULE).sv tb_$(MODULE).cpp
	@echo
	@echo "### VERILATING ###"
	verilator --timing -Wall --trace-fst  -cc $(MODULE).sv --exe tb_$(MODULE).cpp
	@touch .stamp.verilate

.PHONY:lint
lint: $(MODULE).sv
	verilator --lint-only $(MODULE).sv

.PHONY: clean
clean:
	rm -rf .stamp.*;
	rm -rf ./obj_dir
	rm -rf ./logs

```

wsnyder, 2024-09-18:
Please try the most recent version.

wsnyder, 2024-09-18:
```
    VerilatedFstC * m_trace = new VerilatedFstC;
    Vexample * dut = new Vexample; 
```

Opening the trace is when the timescale is determined.  Swap the lines that new() the trace and new() the model, or just use $dumpfile and remove tracing code from your .cpp file.