verilator/verilator #5460: Access qualifier for `typedef` inside class does not work

chykon, 2024-09-18:
Hello!

Verilator allows adding an access qualifier to a `typedef` inside a class, but does not react in any way if it is violated.

The expected behavior is one of two things:

* PREFERRED (standard?): prohibit adding access qualifier - as far as I understand, section "8.18 Data hiding and encapsulation" (IEEE 1800-2023) allows access qualifiers to be applied only to class properties and methods
* ALTERNATIVE (non-standard?): Verilator throws an error when access is violated

**Example:**

```systemverilog
module top;
  class cls;
    local typedef bit t1;
    protected typedef bit t2;
  endclass

  cls::t1 var1; // access error expected but there is none
  cls::t2 var2; // access error expected but there is none
endmodule
```

**Output:**

Normal output without error. Nothing interesting :)

**Command:**

```bash
verilator --binary top.sv
```

---

Verilator version (git master): `Verilator 5.029 devel rev v5.028-88-g195a66529`

OS (verilator devcontainer): Ubuntu 22.04.4 LTS

wsnyder, 2024-09-18:
Three simulators including Verilator allow this, two others give syntax errors. By my reading of IEEE this is legal:

```
type_declaration ::=
          yTYPEDEF ...

data_declaration ::=
         ...
         | type_declaration

class_property ::=
         { property_qualifier } data_declaration

property_qualifier {footnote10} ::=
         random_qualifier
       | class_item_qualifier

class_item_qualifier {footnote10} ::= ySTATIC | yPROTECTED | yLOCAL

Footnote 10) In any one declaration, only one of yPROTECTED or yLOCAL is
allowed, only one of yRAND or yRANDC is allowed, and ySTATIC and/or
yVIRTUAL can appear only once.
```

Verilator checks these properties on non-typedefs, perhaps you'd be willing to make a pull to copy in the warning code appropriately into V3LinkDot's typedef han$

chykon, 2024-09-23:
It seems that this problem is also related to the possibility of using such constructions:

```systemverilog
module top;

  class cls;
    local protected virtual static const rand randc typedef bit bool_t; // something is wrong
  endclass

  initial begin
    cls::bool_t boolean = 1;
  end

endmodule
```

closed 2024-10-06 by wsnyder