verilator/verilator #5461: Error accessing `typedef` declared inside parameterized class

chykon, 2024-09-18:
Hello!

If you access a type declaration (`typedef`) in one parameterized class and use it to specify the type of a parameter of another parameterized class, an error occurs.

The expected behavior is no errors.

Related (maybe) issue: https://github.com/verilator/verilator/issues/4924 (see `Test 2` output)

**Example:**

```systemverilog
module top;

  class cls1#(bit param);
    typedef bit bool_t;
  endclass

  class cls2#(cls1#()::bool_t param); // note that for some reason the parameter for `cls1` may be omitted here
  endclass

  cls2#(1) mycls; // here the parameter for `cls2` cannot be omitted - there will be an error

endmodule
```

If `cls1` is made a simple (non-parameterized) class, then everything works:

```systemverilog
module top;

  class cls1;
    typedef bit bool_t;
  endclass

  class cls2#(cls1::bool_t param);
  endclass

  cls2#(1) mycls;

endmodule

```

**Output:**

```text
%Error: Internal Error: top.sv:7:24: ../V3AstNodeDType.h:1191: Typedef not linked
                                   : ... note: In instance 'top'
    7 |   class cls2#(cls1#()::bool_t param);  
      |                        ^~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

**Command:**

```bash
verilator --binary top.sv
```

---

Verilator version (git master): `Verilator 5.029 devel rev v5.028-88-g195a66529`

OS (verilator devcontainer): Ubuntu 22.04.4 LTS

wsnyder, 2024-09-19:
This error is from V3Scope, it appears related to creating the VDynScope logic (CC @kbieganski).

@chykon might you look into what's going wrong? Basically when V3Scope flattens the design it can't find the given var and scope to make a Varscope.

wsnyder, 2024-09-19:
Ignore that comment - wrong issue.

chykon, 2024-09-19:
Changed the original example with the error a bit (just wrapped `cls1` in a typedef and used it in the parameters of `cls2`) and got a different error.

**Example:**

```systemverilog
module top;

  class cls1#(bit param);
    typedef bit bool_t;
  endclass

  typedef cls1 cls1_t;
  class cls2#(cls1_t#()::bool_t param);
  endclass

  cls2#(1) mycls;

endmodule
```

**Output:**

```text
%Error: a.sv:8:26: syntax error, unexpected IDENTIFIER, expecting TYPE-IDENTIFIER
    8 |   class cls2#(cls1_t#()::bool_t param);
      |                          ^~~~~~
%Error: Internal Error: a.sv:1:8: ../V3ParseSym.h:126: Symbols suggest ending CLASS 'cls2' but parser thinks ending MODULE 'top'
    1 | module top;
      |        ^~~
```