verilator/verilator #5462: Interface Task unpacking: Can't locate varref scope

neeldugar, 2024-09-18:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```systemverilog
interface iface #(parameter DWIDTH = 32)();
  localparam TOTAL_PACKED_WIDTH = DWIDTH + 1;
  modport Tx(output sop, data, import unpack);
  logic sop;
  logic [DWIDTH - 1:0]  data = '0;

  task static unpack(input logic [TOTAL_PACKED_WIDTH-1:0] packed_in, input logic sop_i);
    logic sop_nc;
    {data, sop_nc} <= packed_in;
    sop <= sop_i;
  endtask
endinterface

module t;
iface ifc();
endmodule
```
What 'verilator' command line do we use to run your example?
`verilator --binary -j 0 -Wall foo.sv`

What 'verilator --version' are you using?  Did you try it with the git master version?
`Verilator 5.028 2024-08-21 rev conda-forge build 0`. No.
What OS and distribution are you using?
Linux x86
May we assist you in trying to fix this in Verilator yourself?
Sure

wsnyder, 2024-09-19:
Likely related to __VDynScope_ creation - CC @kbieganski who may need to consult.

@neeldugar might you look into why?  The error is because V3Scope can't locate the variable under the expected scope when trying to flatten the design in V3Scope.cpp.

godblesszhouzhou, 2024-10-20:
I have done some debugs for this issue, a simple case can be:
`module foo (A, B, CIN, S, COUT);
    input A, B;
    input CIN;
    output [1:0]S;
    output COUT;
    reg i;
    reg [1:0]S;
    reg COUT;
    task ADD;
        input A, B, CIN;
        output [1:0] C;
        reg [1:0] C;
        reg S, COUT;
        reg sop_nc;
        reg sop_i;

        begin
            **sop_nc <= A;**
            sop_i <= foo.i;
            S = A ^ B ^ CIN;
            COUT = (A&B) | (A&CIN) | (B&CIN);
            C = {COUT, S};
        end
    endtask
    always @(A or B or CIN) begin
        ADD (A, B, CIN, S);
        COUT = S[0] + S[1];        
    end
endmodule`

Wiht command: `verilator --lint-only --timing foo.sv`

The main problem is that for a delay assign of local var in task,  DynScopeVisitor will replace the delay assign `sop_nc <= A; ` with a new built fork, and the corresponding delay assign will be moved into the fork. Then, as the var of varref A is in the scope of task ADD, in the new built task scope there is not a definition of var A.  That's the reason. So I think, if the RHS of the delay assign exists any input signals, it should not be processed in this method. A simple modification can be, for  DynScopeVisitor

`    void visit(AstAssignDly* nodep) override {
        if (m_procp && !nodep->user2()  // Unhandled AssignDly in function/task
            && nodep->lhsp()->exists(  // And writes to a local variable
                [](AstVarRef* refp) {
                    return refp->access().isWriteOrRW() && refp->varp()->isFuncLocal();
                })
            **// Check whehter RHS contains input Port.
            && nodep->rhsp()->exists(
                [](AstVarRef* refp) {
                    if (refp->access().isReadOrRW() && refp->varp()->isInput()) return false;
                    return true;
                })) {**
            nodep->user2(true);
            // Put it in a fork to prevent lifetime issues with the local
            AstFork* const forkp = new AstFork{nodep->fileline(), "", nullptr};`

wsnyder, 2024-10-20:
Thanks for debugging, can youmake a pull request with the test and fix please?

godblesszhouzhou, 2024-10-27:
[https://github.com/verilator/verilator/pull/5568](url)

wsnyder, 2024-11-02:
Thanks for fixing!

closed 2024-11-02 by wsnyder