verilator/verilator #5463: Question about Toggle Coverage Count with --coverage-toggle

figuresudo, 2024-09-19:
I am currently using the --coverage-toggle option to collect coverage data, and I am able to retrieve the results, including signals and count information. However, I would like to understand how this count is calculated. Specifically, how can I determine whether a signal has transitioned from 1 to 0 and from 0 to 1 based on the collected counts?

For reference, I am using Verilator version 5.020.

wsnyder, 2024-09-19:
It counts every time the signal toggles, that is 0->1, 1->0.  So a count of 1 means it must have been at least 0 and 1 once.

closed 2024-09-19 by wsnyder

figuresudo, 2024-09-19:
So, do you mean that a count of 2 means it must have encountered 0-&gt;1 one time and 1-&gt;0 one time?

wsnyder, 2024-09-19:
Yes, or 1->0 then 0->1.