verilator/verilator #5468: Can't create a VCD

jordankrim, 2024-09-19:
I am running Verilated  code of the CEP.  For version 4.4 it works fine to create a VCD file by running the binary as such:

./simulator-chipyard-CEPVerilatorRocketConfig -v ../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/aesMacro.vcd ../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/riscv_wrapper.elf +loadmem=../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/riscv_wrapper.hex +loadmem_addr=80000000

With both versions 4.5 and 4.7 of the CEP when trying to get a VCD file I get this:

./simulator-chipyard.harness-CEPVerilatorRocketConfig -v ../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/aesMacro.vcd ../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/riscv_wrapper.elf +loadmem=../../sims/cep_cosim/testSuites/bareMetalTests/aesMacro/riscv_wrapper.hex +loadmem_addr=80000000
./simulator-chipyard.harness-CEPVerilatorRocketConfig: invalid option -- 'v'
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Unknown argument (did you mean to enable +permissive parsing?)

wsnyder, 2024-09-19:
That message is coming from whomever created that executable/main() sources, not Verilator, so a question for them.

closed 2024-09-19 by wsnyder

jordankrim, 2024-09-19:
That's what I thought, but needed to try.  Unfortunately, I can't get their attention.

wsnyder, 2024-09-19:
Sorry that's the case.  I presume it's also an open source project and like all of us they don't have enough time. I'd encourage you to look at their sources and try to debug it yourself.