verilator/verilator #5470: FST trace code asserts when model name is empty

sam-shahrestani, 2024-09-20:
I've also experienced the issue in https://github.com/cocotb/cocotb/issues/4074, but it's possible to trigger this with reasonable Verilator-only code, so I think a Verilator issue should track this. A possible fix is https://github.com/verilator/verilator/pull/5313.

### TLDR
Constructing a model with `""` as the name results in no prefix being added in `trace_init` as of https://github.com/verilator/verilator/issues/5261. This causes `assert(hierarchicalName.rfind(' ') != std::string::npos)` to fail when declaring variables for FST trace files.

### Example
This is a small example that doesn't use cocotb at all and triggers the issue on Verilator 5.028:

In particular note:

- The model is constructed as `new Vexample("")` (note that cocotb uses `new Vtop("")` which connects the issues)
- The Verilator command line uses nothing particularly fancy: `-cc --build --exe --trace --trace-fst`
- An empty name is legal according to an emitted comment in the model header:
```
    /// If name is "", then makes a wrapper with a
    /// single model invisible with respect to DPI scope names.
```
Sorry if this is considered a duplicate issue, but again, I couldn't see anything tracking this against Verilator specifically.

sam-shahrestani, 2024-09-20:
Should have double checked my post, the attached example went AWOL somewhere: [example.zip](https://github.com/user-attachments/files/17071042/example.zip)

wsnyder, 2024-09-20:
Makes sense, looks like you're most of the way to a patch, might you be able to convert this to a pull request along with a fix?

wsnyder, 2024-09-21:
Ignore last comment - wrong bug!  Working on this.

closed 2024-09-21 by wsnyder