verilator/verilator #5472: `program` block does not terminate simulation

chykon, 2024-09-20:
Hello!

Simulation does not end after all `initial` processes in the `program` block have completed.

Expected behavior is to terminate the simulation as described in IEEE 1800-2023 section "24.3 The program construct":

---

[image]

---

**Example:**

```systemverilog
program top;
  initial begin
    $display("hello world");
  end
endprogram
```

**Output:**

```text
...
hello world
```

**Command:**

```bash
verilator --binary top.sv && obj_dir/Vtop
```

---

Verilator version (git master): `Verilator 5.029 devel rev v5.028-88-g195a66529`

OS (verilator devcontainer): Ubuntu 22.04.4 LTS

wsnyder, 2024-09-20:
Verilator didn't used require --timing, and so it uses the presence of delay statements to decide if to loop until $finish or until new events.  I don't want to break backwards compatibility, let me think about solutions...

Meantime use:
```
    #1;
    $display("hello world");
```

wsnyder, 2024-09-20:
Or of course add a `$finish`.

wsnyder, 2024-09-20:
Shockingly, the trivial change of checking for timing mode worked.  Fixed.

closed 2024-09-20 by wsnyder

chykon, 2024-09-21:
Great job! Just a small note about the [test code](https://github.com/verilator/verilator/commit/92dd8ee8f5a456b9b27fc162e3ffbe84e0592333#diff-d35f4e9871f75ec07253cb66f95a00c0e035c9d62a22b8890adad41882ea2960R9):

```systemverilog
program t;
   initial begin
      #1;
      $write("*-* All Finished *-*\n");
      // No $finish
   end
endprogram
```

Is the delay `#1` in the test code intentional? It is not needed after the fix.