verilator/verilator #5480: The simulation terminates after the `initial` process in the module has completed.

chykon, 2024-09-22:
Hello!

It appears that after fixing #5472 there was a regression and now the simulation stops after executing the `initial` process even in a `module` block (not only in `program` block).

Expected behavior is that the simulation does not terminate after the `initial` process in the module block has completed.

**Example:**

```systemverilog
module top;
  initial begin
    $display("hello");
    // no $finish
  end
endmodule
```

**Output:**

```text
hello
- S i m u l a t i o n   R e p o r t: Verilator 5.029 devel
- Verilator: end at 0s; walltime 0.000 s; speed 0.000 s/s
- Verilator: cpu 0.000 s on 1 threads; alloced 8 MB
```

**Command:**

```bash
verilator --binary top.sv && obj_dir/Vtop
```

---

Verilator version (git master): `Verilator 5.029 devel rev v5.028-126-g2584befa5`

OS (verilator devcontainer): Ubuntu 22.04.4 LTS

wsnyder, 2024-09-22:
That's intentional and my reading of IEEE, where do you see language to contrary?

wsnyder, 2024-09-22:
Confirmed modules exit on all other simulators.

closed 2024-09-22 by wsnyder

chykon, 2024-09-22:
Yes, the current behavior is correct, sorry.

If someone finds this issue, see IEEE 1800-2023 "4.5 SystemVerilog simulation reference algorithm". The `execute_simulation` procedure will describe the current behavior.