verilator/verilator #5484: Fix Short Circuiting with Associative Array Accesses

sifferman, 2024-09-24:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.028-135-g5a65d2785
```

Hello. I discovered a bug where short circuiting does not stop elements from being added to an associative array. (Although it is possible this bug is not specific to associative arrays.)

```systemverilog
module tb;

logic [31:0] dict [int];

// verilator lint_off WIDTHTRUNC
function automatic logic f(int a);
    int dict_size = dict.size;
    logic next_exists = dict.next(a);

    // incorrectly inserts element at `a`
    logic next_nonzero = !next_exists || (dict[a] != 0);
    assert (dict_size == dict.size) else begin
        $display("Assertion failed: dict_size mismatch");
        $display("Initial size: %0d, New size: %0d", dict_size, dict.size);
        $display("Dictionary contents:");
        foreach (dict[key]) begin
            $display(" Key: %0d, Value: %0d", key, dict[key]);
        end
        $error;
    end

    return next_nonzero;
endfunction

initial begin
    logic r = f(0);
end

endmodule
```
Output:
```
Assertion failed: dict_size mismatch
Initial size: 0, New size: 1
Dictionary contents:
 Key: 0, Value: 0
[0] %Error: tb.sv:20: Assertion failed
```
I won't have time to look into this for awhile.

wsnyder, 2024-09-24:
This passes for me, what is the command line, and are you sure you are using master?

sifferman, 2024-09-24:
Yes, I am using master.

Command:
```
verilator tb.sv --binary --assert
```

Maybe it's a compiler issue? I have
```
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
```

wsnyder, 2024-09-24:
My mistake, forgot the --assert.

Simple, fixed.

closed 2024-09-24 by wsnyder

sifferman, 2024-09-24:
Thanks so much!