verilator/verilator #5485: Fix non-blocking assignments at end of always block

sifferman, 2024-09-24:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.026-284-g4ec75b261
```

Hello. I found a bug where non-blocking assignments don't work as intended at the end of an always block.

```systemverilog
// tb.sv
module tb;

int delay = 0; always #1 delay = int'($time) / 4;
task automatic do_delay;
    if (delay > 0) #(delay);
endtask

// `a` should match `b`
// a runs correctly
logic a = 0;
always begin
    a <= 1;
    do_delay();
    a <= 0;

    #1;
end

// b runs incorrectly
logic b = 0;
always begin
    b <= 1;
    do_delay();
    b <= 0;

    #1;
    b <= 1; // this line should do nothing
end

initial begin
    $dumpfile("dump.vcd");
    $dumpvars;
    #20;
    $finish;
end

endmodule
```
```
// verilator.f
tb.sv
--binary
--trace
```

In simulation, `a` does not match `b` when the `do_delay()` task does not create a delay:

[image]

I won't have time to look into this for awhile.

wsnyder, 2024-09-24:
Would you mind updating the test to be self-checking?  Thanks.

sifferman, 2024-09-24:
Done

wsnyder, 2024-09-24:
@kbieganski mind taking a look when get a chance? Thanks.

wsnyder, 2024-09-24:
`test.compile(verilator_flags2=['--binary --assert'])
`

rhanqtl, 2024-10-04:
I added a `$monitor` call in the initial block before `#20` and found that Verilator's simulation result is different from other popular simulators (using EDA Playground):

| Simulation Time | Synopsys VCS | Cadence Xcelium | Icarus Verilog | Verilator |
|---:|---|---|---|---|
| 0 | a=0 b=0 | a=0 b=0 | a=0 b=0 | a=0 b=0 | 
| 1 | || a=0 b=0 (?) | a=0 b=1 |
| 2 | || a=0 b=0 (?) | |
| 3 ||| a=0 b=0 (?) | |
| 4 | a=1 b=1 | a=1 b=1 | a=1 b=1 | |
| 5 | a=0 b=0 | a=0 b=0 | a=0 b=0 | a=1 b=1 |
| 6 | a=1 b=1 | a=1 b=1 | a=1 b=1 | a=0 b=0 |
|7| a=0 b=0 | a=0 b=0 | a=0 b=0 | a=1 b=1 |
|8| a=1 b=1 | a=1 b=1 | a=1 b=1 | a=0 b=0 |
|9| ||| a=1 b=1 |
|10| a=0 b=0 | a=0 b=0 | a=0 b=0 |
|11| a=1 b=1 | a=1 b=1 | a=1 b=1 | a=0 b=0 |
| 12 | ||| a=1 b=1 |
|13| a=0 b=0 | a=0 b=0 | a=0 b=0 | |
|14| a=1 b=1 | a=1 b=1 | a=1 b=1 | a=0 b=0 |
| 15 | ||| a=1 b=1 |
| 16 | |||
|17| a=0 b=0 | a=0 b=0 | a=0 b=0 | |
|18| a=1 b=1 | a=1 b=1 | a=1 b=1 | a=0 b=0 |
| 19 |||| a=1 b=1 |

It seems that if the `a=0 b=1` at time 1 of Verilator is removed, everything will be OK.

gezalore, 2024-10-04:
I'm doing some major work in V3Delayed right now that might or might not help this. Just in case someone is working on it please let me know.

gezalore, 2025-09-06:
Yeah, that was the rework to support NBAs to arrays in loops. I can have a look in a short while.

closed 2026-11-05 by gezalore