verilator/verilator #5486: Fix Short Circuiting with Associative Arrays of Dynamic Arrays

sifferman, 2024-09-24:
```
Ubuntu 22.04.4 LTS
Verilator 5.029 devel rev v5.028-137-g811eab8fa
```

Hello. It looks like #5484 wasn't completely fixed. The problem still exists for associative arrays of dynamic arrays.

```systemverilog
//tb.sv
module tb;

logic [31:0] dict [int] [];

// verilator lint_off WIDTHTRUNC
function automatic logic f(int a);
    int dict_size = dict.size;
    logic next_exists = dict.next(a);

    // incorrectly inserts element at `a`
    logic next_nonempty = !next_exists || (dict[a].size != 0);
    assert (dict_size == dict.size) else begin
        $display("Assertion failed: dict_size mismatch");
        $display("Initial size: %0d, New size: %0d", dict_size, dict.size);
        $display("Dictionary contents:");
        foreach (dict[key]) begin
            $display(" Key: %0d, Value Size: %0d", key, dict[key].size);
        end
        $error;
    end

    return next_nonempty;
endfunction

initial begin
    logic r = f(0);
end

endmodule
```
```
// verilator.f
tb.sv
--binary
--assert
```
Output:
```
Assertion failed: dict_size mismatch
Initial size: 0, New size: 1
Dictionary contents:
 Key: 0, Value: 0
[0] %Error: tb.sv:20: Assertion failed
```

wsnyder, 2024-09-25:
FWIW it's related to the .size method call, not the use of a dynamic array.

closed 2024-09-25 by wsnyder

sifferman, 2024-09-25:
Thanks! The fix works perfectly