verilator/verilator #5487: Scheduling reflection of program instances.

katsui97, 2024-09-25:
The reflection of program instances in SystemVerilog is specified to occur in the Reactive region, but in Verilator, it seems to be reflected on the Active side instead. There is no issue with clocking blocks as they are correctly executed in the Reactive region, but I am curious about how program instances are implemented. Since program cannot be used with UVM, it might be acceptable to drop it from the specification, but as it stands, it is somewhat partially supported, which feels incomplete. In VCS and Xcelium, it reflects at the timing of the Reactive region.

Verilator 5.029 devel rev v5.028-136-g4ec75b261
------------------------------------------------------------------------------
`timescale 1ns/100ps

module top();

  // VCD or FST Dump
  initial begin
    $dumpfile("wave.dump");
    $dumpvars(0, top);
  end

  bit RST_X = 1'b1;

  bit CLK;
  always #10
    CLK = ~CLK;

  logic [3:0] r_cnt;
  always_ff @(posedge CLK, negedge RST_X)
    if(~RST_X)
      r_cnt <= '0;
    else
      r_cnt <= 4'(r_cnt + 1'b1);

  clocking cb @(posedge CLK);
  endclocking

  event Ev_rtl_tmg, Ev_cb_tmg, Ev_prg_tmg;;
  always_ff @(posedge CLK)
    if(r_cnt == 4'd5)
      ->Ev_rtl_tmg;       // RTL (Active region)

  initial begin
    #3  RST_X = 1'b0;
    #15 RST_X = 1'b1;

    repeat (30) begin
      @(cb);

      if(r_cnt == 4'd5)
        ->Ev_cb_tmg;   // Cloking block (Reactive region)
    end

    @(posedge CLK);
    $finish;
  end

  test_prg  test_prg();   // Program instance

endmodule

program test_prg();

  initial
    forever
    begin
      @(posedge top.CLK);
      if(top.r_cnt == 4'd5)
        ->top.Ev_prg_tmg;
    end
endprogram

[image]

wsnyder, 2024-09-25:
Programs currently alias to modules, as nothing has broken until now ;)

Aside from "testing IEEE", what real framework is making use of programs and needs this behavior?  If there is one, we'd welcome  a pull request to add the related regions, should be fairly straight forward.  I think it's unlikely that we'd get to this soon otherwise.

katsui97, 2024-09-25:
Thank you for your response.
If there are no requests from others, I think it's fine not to make any changes.
It’s more of a request that, if possible, I would like it to be addressed.
I was unaware of the existence of clocking blocks and thought that the program was the only reactive behavior.
Since my company uses commercial simulators for work, this doesn't affect us.

wsnyder, 2024-09-25:
Thanks, I'll just close this for now then as our work list is too long ;), if others find this is a problem we can reopen.

closed 2024-09-25 by wsnyder