verilator/verilator #5490: Partially out of range select is not randomized

gezalore, 2024-09-26:
Given
```verilog
module top(
  input wire logic [2:0] i,
  output wire logic[3:0] o
);
  localparam [4:0] p = 5'h1f;
  assign o = p[i +: 4];
endmodule
```

With `verilator --cc --x-assign unique`, we generate the following due to V3Unkown:

```c++
       vlSelfRef.o = (0xfU & ((4U >= (IData)(vlSelfRef.i))
                               ? (0x1fU >> (IData)(vlSelfRef.i))
                               : (IData)(vlSelfRef.top__DOT____Vxrand_h8d94a835__0)));
```

This fails to randomize the out of range bits when `i` is 2/3/4.

Mainly filing this for reference in upcoming PR for another bug.

wsnyder, 2024-09-26:
This is an intentional bug (for better and worse), as I didn't have a decent-performing alternative idea.

Note also #2819 which is a related select-performance-hack problem.

gezalore, 2024-09-26:
Maybe an idea: Extend the selected variable by <width of widest variable select>-1, and randomize the top bits.

Probably doesn't work for top level io, but you could rewrite:

```verilog
   logic [9:0] a;
   wire logic [3:0] x = a[i+:3];
```
into 
```verilog
   logic [12:0] a;
   wire logic [3:0] x = i <= 9 ? a[i+:4] : AstRand;
   initial a[12:10] = AstRand;
```

Probably doesn't work for top-level IO, and maybe for others either.

wsnyder, 2024-09-26:
Also this logic predates the option to select x-randomization on/off so it isn't as critical as when written.