verilator/verilator #5498: If the simulation is forcibly terminated, waveform capture in FST mode is not possible.

katsui97, 2024-10-01:
It seems that when outputting waveforms in FST mode and forcibly terminating the simulation, the waveform cannot be loaded into GTKWave.
On the other hand, in VCD mode, the waveform can be captured up to a certain point without any issues.

---------------------------------------------------------
GTKWave Analyzer v3.3.104 (w)1999-2020 BSI
GTKWAVE | Could not initialize 'wave.dump', exiting.
---------------------------------------------------------

Is it possible to support this by default without requiring the user to modify the source code?

Verilator 5.029 devel rev v5.028-155-g13a124035

wsnyder, 2024-10-01:
Terminating meaning SIGTERM signal? I suppose we could catch and flush buffers, but it has side risks and will never completely avoid this problem.

I'd suggest instead GTK should instead be more tolerant of truncated input? Maybe with a warning.

katsui97, 2024-10-01:
Yes, it's either SIGTERM or SIGINT.
Since the FST format is compressed binary, unlike the simple output of VCD, does this mean that just flushing the buffer and closing the file won't work properly?

wsnyder, 2024-10-01:
Probably not in all cases, which is why seems better to have GTK robustly accept truncations.

katsui97, 2024-10-01:
Thank you very much. Understood. Since there seems to be some risks involved with doing this in Verilator, please feel free to close this issue if it seems too difficult. While Verilator is being actively developed, GTKWave seems to be more stable, with no major changes recently. It seems that I haven’t pulled the latest version from Git, so it’s possible that the issue might already be addressed.

closed 2024-10-01 by wsnyder