verilator/verilator #5499: There is no reaction to the edge event at zero time

punzik, 2024-10-01:
The `always @(posedge a)` block does not triggered on change of `a` in zero time.

The SV standard says that the "initial value of a static variable as part of the variable declaration (including static class
members) shall occur before any initial or always procedures are started." (chap. 10.5).

Example code:
```verilog
module test;
    logic a = 1'b0;
    logic b = 1'b0;

    always @(posedge a) b <= 1'b1;

    initial begin
        a = 1'b1;
        #1 $display("b = %b", b);
        $finish;
    end

endmodule // test
```

Expected output:
```
b = 1
```

Actual:
```
b = 0
```

Verilator version: 03012da11c54f614154400b01a12ad30ceeb8a30

wsnyder, 2024-10-01:
Note this is different than X->0 edge sensitivity, so I'd change the example to:

```
module t;
    bit a = 1'b0;
    bit b = 1'b0;

    always @(posedge a) b <= 1'b1;

    initial begin
        a = 1'b1;
        #1;
        $display("b = %b", b);
        if (b !== 1) $stop;
        $write("*-* All Finished *-*\n");
        $finish;
    end
endmodule
```

wsnyder, 2024-10-01:
Presumably at init time we need to add all of the appropriate activations. This is annoying as might add overhead that slows down post-init simulations; an unfortunate IEEE decision.

@kbieganski thoughts?

@punzik might you work on a pull to fix this?

commit 4a221294 (2025-03-18): Fix change detection at time 0 (#5864)

Initialize "previous value" variables in the static initializer
function, instead of the 'initial' blocks function. Fixes #5499