verilator/verilator #5507: CMake: verilate command bugs

Febbe, 2024-10-04:
Version <= git master 
OS <= windows 11
cmake <= 3.23.0.rc5

I have several issues with the verilate command in cmake:

The primary model header is named after the first source in the `SOURCES` list. This is intuitive, when the toplevel is not defined first.
The solution would be, to first identify the toplevel (optionally all toplevels) and then generate header files for each toplevel.
When `TOP_MODULE` is defined, use that identifier.

The PREFIX option causes compilation errors on windows, when `".."` as path is used. The cause is that somehow classes in header and sourcefiles are named `class ..`, note the two dots. This is a forbidden char for identifiers in C++.
The solution to this, would be to get the canonical path and use the foldername. Also prepend some characters like `ver_` and replace forbidden chars with `_`.

wsnyder, 2024-10-04:
1. The top level .v should be given to Verilator first in SOURCES, not sure if anydocumentation or Cmakefile change is needed?

2. prefix is supposed to be given only as a legal identifier, not a path.

Note Verilator does not yet fully support Windows filenames. #4255

Febbe, 2024-10-04:
> 1. The top level .v should be given to Verilator first in SOURCES, not sure if anydocumentation or Cmakefile change is needed?
> 
> 2. prefix is supposed to be given only as a legal identifier, not a path.
> 
> Note Verilator does not yet fully support Windows filenames. #4255 

For 1. It would be more clear, to support arbitrary orders of files.

For 2. the documentation may be improved I thought prefix should be a path, not an identifier, also an error message, that the input is not valid should be raised, before it is passed to the code generation.

wsnyder, 2024-10-04:
1. sources can be arbitrary order if you give a prefix.  Using GNU make it is a bit more flexible as it computes prefix based on finding the top module.
2. Agreed; might you consider a pull to check validity and improve docs? Thanks.

Febbe, 2024-10-04:
> 1. sources can be arbitrary order if you give a prefix.  Using GNU make it is a bit more flexible as it computes prefix based on finding the top module.
> 2. Agreed; might you consider a pull to check validity and improve docs? Thanks.

I would consider improving both, but I have to do this in my free time and I have several other projects (xournalpp, clang-query support in clangd, Bluespec support on windows). The previous PR was only a result of the steps I did, to compile verilator for windows at my workplace.

closed 2024-11-27 by wsnyder