verilator/verilator #5509: Error when using a for loop variable in a slice range

jeras, 2024-10-04:
Using the latest git build. I attached the SystemVerilog code causing the issue.

[loop_variable_in_range.txt](https://github.com/user-attachments/files/17259786/loop_variable_in_range.txt)

I would expect `for` loop variables to be seen as constants within each loop iteration, instead I get an error. The code compiles well in Questa (I only tried with the original code this example was made from). This is the Verilator output:

```
%Error: loop_variable_in_range.sv:37:32: Expecting expression to be constant, but variable isn't const: 'i'
                                       : ... note: In instance 'loop_variable_in_range'
   37 |             thr_b[i] = |pry[0+:i+1];
      |                                ^
%Error: loop_variable_in_range.sv:37:33: Width of :+ or :- bit slice range isn't a constant
                                       : ... note: In instance 'loop_variable_in_range'
   37 |             thr_b[i] = |pry[0+:i+1];
      |                                 ^
%Error: loop_variable_in_range.sv:41:29: First value of [a:b] isn't a constant, maybe you want +: or -:
                                       : ... note: In instance 'loop_variable_in_range'
   41 |             thr_c[i] = |pry[i:0];
      |                             ^
```

wsnyder, 2024-10-04:
IEEE:

```
indexed_range ::=
         expression '+:' constant_expression
```

That is in your example, `i` must be a genvar or parameter.  As it is not, this is a Questa bug.

closed 2024-10-04 by wsnyder

jeras, 2024-10-04:
Thanks. I tried to run the example on EDAplayground, using 4 different simulators and they all returned an error. I am not sure why this seemed to work before in questa, but it does not really matter. I will rewrite the code.

jeras, 2024-10-05:
I am researching various synthesis primitives and I would like some **feedback on the syntax**. In this case the code represents a [prefix sum](https://en.wikipedia.org/wiki/Prefix_sum) OR/AND/XOR operation. The most common operation implemented as a prefix sum are binary adders where the carry propagation can be implemented as a chain ([ripple carry adder](https://en.wikipedia.org/wiki/Adder_(electronics)#Ripple-carry_adder)) or a parallel network (there are many solutions with different compromises, the most common are [Brent–Kung and Kogge–Stone](https://en.wikipedia.org/wiki/Brent%E2%80%93Kung_adder#Comparison_with_Kogge%E2%80%93Stone_adder). While adders have an obvious syntax using the `+` operator, other even simpler operations do not.

Prefix sum OR/AND operations can be used in incrementers, priority encoders,... Prefix sum XOR is used in Gray to binary encoding. I will use a priority encoder (LSB has highest priority) as an example. The priority encoder would convert a priority vector into a one-hot vector, where the hot bit is the least significant active bit in the priority vector.

As an example the priority vector `pry[8-1:0] = 8'b0101_1000` would be converted onto the one-hot vector `oht[8-1:0] = 8'b0000_1000`. This can be done in two steps, first a prefix sum OR (consuming most of the area/timing) to convert to a thermometer encoding `thr[8-1:0] = 8'b1111_1000` followed by a simple bitwise operation `oht[8-1:0] = thr[8-1:0] & {thr[8-2:0], 1'b0}`.

I tried to simulate and synthesize different syntax-es for the priority to thermometer encoding trying to find one which simple, so it is easy to write without bugs and could be recognized by synthesis tools so they could implement it as a ripple chain or prefix sum network depending on timing requirements. Yosys does not recognize this prefix sum structures as primitives (except for arithmetic operations), but some future version might.

I will start with a syntax that is recognized by all simulation/sysnthesis tools I tried. The obvious downside of this syntax is its verbosity, the temporary variable `tmp` is there just to avoid tools complaining about combinational loops.

```SystemVerilog
always_comb
begin: loop
    logic [WIDTH-1:-1] tmp;
    tmp[-1] = 1'b0;
    for (int i=0; i<WIDTH; i++) begin
        tmp[i] = pry[i] | tmp[i-1];
    end
    thr = tmp[WIDTH-1:0];
end: loop
```

The same could be written without the temporary signal but I get complaints about combinational loops.

```SystemVerilog
                    thr[0] = pry[0];
                    for (int i=1; i<WIDTH; i++) begin
                        thr[i] = pry[i] | thr[i-1];
                    end
```

```sh
%Warning-ALWCOMBORDER: ../../rtl/logic/pry2thr_base.sv:36:29: Always_comb variable driven after use: 'thr'
                                                            : ... note: In instance 'pry2thr_tb.imp[0].dut.exact.pry2thr_tree.branch.pry2thr_sub.leaf.pry2thr'
   36 |                             thr[i] = pry[i] | thr[i-1];
      |                             ^~~
                       ... For warning description see https://verilator.org/warn/ALWCOMBORDER?v=5.029
                       ... Use "/* verilator lint_off ALWCOMBORDER */" and lint_on around source to disable this message.
```

Even if I silence the `ALWCOMBORDER` warning, I still get `UNOPTFLAT`:
```sh
 %Warning-UNOPTFLAT: ../../rtl/logic/pry2thr_tree.sv:44:45: Signal unoptimizable: Circular combinational logic: 'pry2thr_tb.imp[1].dut.exact.pry2thr_tree.branch.thr_brn'
   44 |         logic [SPLIT-1:0]                   thr_brn;   
      |                                             ^~~~~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=5.029
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
```

The most compact syntax would be just a vector operation, but this one might not be accepted by any tools (I did not test it much, so I am not sure).

```SystemVerilog
thr = pry | {thr[WIDTH-2:0], 1'b0};
```

None of this syntax-es is explicit, so it would not be trivial for a synthesis tool to recognize them. But an explicit syntax would almost certainly require a new language feature.

There is another syntax using addition, this one might be even used as an internal implementation in simulators. But it is not appropriate for synthesis tools. Again I am not sure whether I wrote it correctly, it was taken from [Hacker's Delight](https://en.wikipedia.org/wiki/Hacker%27s_Delight#).

```SystemVerilog
thr = pry & -pry;
```