verilator/verilator #5510: SIDEEFFECT warning false positives on coverage

AndrewNolte, 2024-10-04:
I haven't been able to narrow this down to a reproducer yet, but 
- this came up sometime between June and now
- the warnings only show up when building with coverage enabled with `--coverage-line --coverage-user`
- one is a package that reads environment, so not pure but also no sideffect
- the other should be a pure function, but does some concatentations, and is defined within a module

wsnyder, 2024-10-06:
Will need a test case, as you probably already know ;)

wsnyder, 2024-11-27:
Please reopen if get test case.

RRozak, 2025-02-05:
I got a test case:
```systemverilog
module t;
  logic [1:0] buf_byteen;
  logic       cmd_done;

  function automatic logic get_nxtbyte_ptr;
  endfunction

  always begin
    case (1)
      1: begin
        cmd_done = buf_byteen[get_nxtbyte_ptr()];
      end
    endcase
  end
endmodule
```
Command is:
```
verilator --binary --coverage-line <file>
```
Error msg:
```
%Warning-SIDEEFFECT: side_effect.sv:9:31: Expression side effect may be mishandled
                                        : ... note: In instance 'axi4_to_ahb'
                                        : ... Suggest use a temporary variable in place of this expression
    9 |         cmd_done = buf_byteen[get_nxtbyte_ptr()];
      |                               ^~~~~~~~~~~~~~~
                     ... For warning description see https://verilator.org/warn/SIDEEFFECT?v=5.033
                     ... Use "/* verilator lint_off SIDEEFFECT */" and lint_on around source to disable this message.

```

RRozak, 2025-02-05:
`AstCoverInc` node, which is added to a function body when coverage measurement is enabled, is impure. This makes calls to that function impure. Handling of some constructs involves duplication of expressions, which leads to wrong results if these expressions have side effects.

However, this node can be safely duplicated. It can't be removed, can't be added if not present, but duplication won't change the results.

Marking that node as pure is risky, because it may be then removed by optimizations.

I think that we should define a special value for purity, which would mean that the node can be duplicated, but can't be removed, added and there are some restrictions regarding moving it.
This value may be also used for member selects and calls of pure methods, because adding or moving them may result in nullptr dereference during simulation, but duplication of them is safe.

wsnyder, 2025-02-07:
I agree that AstCoverInc is not pure, but you'll get an impure warning on this code if you do nearly anything else in the function. Also note that making it impure means you might get an e.g. count of 2 rather than a count of 1 if it is called what the user things should be once.  So I'm wondering if solving the CoverInc will really help your larger use case?

RRozak, 2025-02-07:
Another approach is to create copies of functions that would have `AstCoverInc` nodes removed. But if such a function calls another function, the called function would also need to have these nodes removed. There are a few phases in which we do conversions which duplicate expressions, so we may end up with many copies of the same function. To prevent from multiple copies we would need to have a global map that would store these copies.

wsnyder, 2025-02-07:
Unfortunately there's a lot of cleanup needed to prevent SIDEEFFECT, which is why the warning exists rather than it just being fixed.