verilator/verilator #5511: crash on array declaration using sizes with initialization assignment pattern of wrong size

jeras, 2024-10-06:
This code causes a crash. It seems (I did not check the waveforms yet) to work correctly if the array and assignment pattern are of the same size.

```SystemVerilog
localparam logic [8-1:0] max_lst [5] = '{0, 1, 2**8-1};
```

The crash output is:

```
- V e r i l a t i o n   R e p o r t: Verilator 5.029 devel rev v5.028-171-gfd9f5ab34
- Verilator: Built from 2.843 MB sources in 59 modules, into 0.017 MB in 5 C++ files needing 0.000 MB
- Verilator: Walltime 0.076 s (elab=0.002, cvt=0.001, bld=0.000); cpu 0.077 s on 16 threads; alloced 55.902 MB
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null;
```

jeras, 2024-10-06:
I noticed, the above code by itself does not cause a crash, instead it reports `Assignment pattern with too many elements`.
I used the array in a `foreach` loop, so this combination might be the issues, I will try to create a minimal example.

jeras, 2024-10-07:
I got a small reproducible example. This is the SystemVerilog code:

```SystemVerilog
module crash ();
    localparam logic [4-1:0] max_lst [5] = '{0, 1, 2**4-1};
endmodule: crash
```

And this is the command I used to compile it:

```bash
verilator --binary --timing --top crash -Wno-INITIALDLY -Wno-UNOPTFLAT --trace-fst crash.sv
```

The output is:

```sh
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/share/verilator/bin/verilator_bin --binary --timing --top crash -Wno-INITIALDLY -Wno-UNOPTFLAT --trace-fst crash.sv
```

wsnyder, 2024-10-07:
Thanks, does this pass on other simulators?  IEEE isn't clear if not initializing everything (or using default) is legal or not.

wsnyder, 2024-10-07:
Confirmed this is illegal.  Verilator does have a check for wrong size (`"Assignment pattern with too many elements"`) so not sure why it isn't triggering.  Might you be able to debug and provide a pull to fix this?

wsnyder, 2024-10-07:
I mean in this case `"Assignment pattern missed initializing elements: "`

jeras, 2024-10-07:
Sorry, I have no experience debugging C++. I do wish to learn and I do have some free time now, but I do not wish to commit to it.

While I did not try with different sizes, I noticed, that when the assignment pattern was larger than the array, I got an error pointing to the line of code, while when the pattern was smaller than the array, I got a crash.

wsnyder, 2024-10-10:
I added an assertion that now fires, but really V3Width needs to detect this. The complication is that array slicing isn't currently computed in V3Width - it currently only checks for constant initializers being correct size.

Better fix welcome.

wsnyder, 2025-09-13:
Fixed in master while ago, example gets:

```
%Error: t/t_0_bad.v:12:37: Assignment pattern missed initializing elements: 3
                         : ... note: In instance 't'
   12 |    localparam logic [7:0] PAR [5] = '{0, 1, 2**8-1};
      |                                     ^~

```