verilator/verilator #5512: Internal error with structure initialization using default with custom type

uge, 2024-10-06:
(thank you for this fantastic tool)

In using Verilator to link some systemverilog, I wanted to initialize some typedef'ed custom types that are buried a few layers of typing down inside a complex structure. Since the type is an enumerated type, I wanted to set all values to the reset value. Verilator threw an internal error. VCS and dc_shell ate it ok. I distilled my code down to the attached test case. This was run on 5.028.

I looked at the LRM to see if this is an acceptable use but couldn't find clear language or examples for nested types.

[verilator_bug.sv.txt](https://github.com/user-attachments/files/17267914/verilator_bug.sv.txt)
[verilator_gdbbt_output.txt](https://github.com/user-attachments/files/17267916/verilator_gdbbt_output.txt)

wsnyder, 2024-10-06:
Need to update fix for clang issue, and make test self-checking.

uge, 2024-10-06:
Confirmed [4b713e9](https://github.com/verilator/verilator/commit/4b713e9882e32869f344c7d6ac4f9b977cbc0dc7) on the internal test case as well. Thank you.