verilator/verilator #5513: Parallel Builds May Lead to DPI-C Linking Errors

Dragon-Git, 2024-10-07:
When I build use default `VM_PARALLEL_BUILDS` make flag, the Verilator run will fail with the following error message.
```
dyld[28472]: symbol not found in flat namespace '_hello'
zsh: abort      ./obj_dir/Vtop
```
But when I build with `-MAKEFLAGS VM_PARALLEL_BUILDS=0`, the Verilator run works fine.

- Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```cpp
// py_dpi.cpp
#include <pybind11/pybind11.h>
#include <pybind11/embed.h>
#include "svdpi.h"

extern "C" {

namespace py = pybind11;

void hello();

PYBIND11_MODULE(py_dpi, m) {
    m.doc() = "svuvm api module";

    m.def("hello", &hello, "suspend the simulation");

}

}
```

```systemverilog
//tpl.sv
package dbg_tpl_pkg;
    class tpl_object#(type T=int, string NAME = "");
        T t;
        function new(string name = NAME);
            this.t = $urandom_range(0, 100);
        endfunction

    endclass

// from 0 to 29
typedef tpl_object#(int, "tpl_object_0") tpl_object_0;
typedef tpl_object#(int, "tpl_object_1") tpl_object_1;
typedef tpl_object#(int, "tpl_object_2") tpl_object_2;
typedef tpl_object#(int, "tpl_object_3") tpl_object_3;
typedef tpl_object#(int, "tpl_object_4") tpl_object_4;
typedef tpl_object#(int, "tpl_object_5") tpl_object_5;
typedef tpl_object#(int, "tpl_object_6") tpl_object_6;
typedef tpl_object#(int, "tpl_object_7") tpl_object_7;
typedef tpl_object#(int, "tpl_object_8") tpl_object_8;
typedef tpl_object#(int, "tpl_object_9") tpl_object_9;
typedef tpl_object#(int, "tpl_object_10") tpl_object_10;
typedef tpl_object#(int, "tpl_object_11") tpl_object_11;
typedef tpl_object#(int, "tpl_object_12") tpl_object_12;
typedef tpl_object#(int, "tpl_object_13") tpl_object_13;
typedef tpl_object#(int, "tpl_object_14") tpl_object_14;
typedef tpl_object#(int, "tpl_object_15") tpl_object_15;
typedef tpl_object#(int, "tpl_object_16") tpl_object_16;
typedef tpl_object#(int, "tpl_object_17") tpl_object_17;
typedef tpl_object#(int, "tpl_object_18") tpl_object_18;
typedef tpl_object#(int, "tpl_object_19") tpl_object_19;
typedef tpl_object#(int, "tpl_object_20") tpl_object_20;
typedef tpl_object#(int, "tpl_object_21") tpl_object_21;
typedef tpl_object#(int, "tpl_object_22") tpl_object_22;
typedef tpl_object#(int, "tpl_object_23") tpl_object_23;
typedef tpl_object#(int, "tpl_object_24") tpl_object_24;
typedef tpl_object#(int, "tpl_object_25") tpl_object_25;
typedef tpl_object#(int, "tpl_object_26") tpl_object_26;
typedef tpl_object#(int, "tpl_object_27") tpl_object_27;
typedef tpl_object#(int, "tpl_object_28") tpl_object_28;
typedef tpl_object#(int, "tpl_object_29") tpl_object_29;

endpackage
```

```SystemVerilog
//top.sv
module top ();
    task hello();
        $display("Hello World!");
    endtask

    export "DPI-C" task hello;
    import "DPI-C" pure function string dirname(string file_path);

    initial begin
        $display(dirname(`__FILE__));
        $finish;
    end
endmodule
```
- What command line do we use to run your example?

```bash
pip3 install pybind11

g++ -O3 -Wall -shared -std=c++17 -fPIC  -undefined dynamic_lookup $(python3 -m pybind11 --includes) py_dpi.cpp -o py_dpi$(python3-config --extension-suffix) $(python3-config --ldflags --embed) -I /opt/homebrew/Cellar/verilator/5.028/share/verilator/include/vltstd

verilator -j $(nproc) --binary -top top tpl.sv tpl_top.sv ../py_dpi$(python3-config --extension-suffix) -LDFLAGS "$(python3-config --ldflags --embed)" 

./obj_dir/Vtop
```

- What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.028 2024-08-21 rev UNKNOWN.REV

- What OS and distribution are you using?

macOS  Sonoma 14.6.1, ubuntu 22.04

wsnyder, 2024-10-07:
When I use a simple .cpp DPI file (without python) and compile on Linux this works fine.  I can't test MacOS, so you'll need to debug this at your end.  It's likely a problem with the Makefile link line, with MacOs, or the special link flags you are using, so for example try doing this without using python libraries.  If you find the issue please report back & make a pull if it's something Verilator can do better.

Dragon-Git, 2024-10-07:
Hi wsnyder

Thank you for your response and suggestions.

I can confirm that the issue only appears after introducing pybind11. Based on my debugging, it seems that the problem is likely related to the order of dependencies when using `ar` in the Makefile. Specifically, it appears that the generation of `verilator_deplist.tmp` might not be taking the dependency order into account.

I will try to reproduce this on an Ubuntu system to further investigate and confirm the root cause. If I find any additional information or a solution, I will report back here.

In the meantime, I will also try to compile without the Python libraries to see if the issue persists. If I can narrow down the problem, I will provide more details and, if necessary, make a pull request to improve Verilator.

Thanks for your help and support.

Dragon-Git, 2024-10-07:
Hi wsnyder,

Thank you for your guidance and patience.

I have completed both tasks:

- I was able to reproduce the issue even without the Python libraries.
- The problem can be  reproduced on both Ubuntu and macOS systems.
Here is a minimal example that demonstrates the issue:

```cpp
// py_dpi.cpp
#include "svdpi.h"
#include <stdio.h>

extern "C" {

void hello();

void hello_wrap() {
    printf("addr: %p", &hello);
}

}
```

```systemverilog
//tpl.sv
package dbg_tpl_pkg;
    class tpl_object#(type T=int, string NAME = "");
        T t;
        function new(string name = NAME);
            this.t = $urandom_range(0, 100);
        endfunction

    endclass

// from 0 to 29
typedef tpl_object#(int, "tpl_object_0") tpl_object_0;
typedef tpl_object#(int, "tpl_object_1") tpl_object_1;
typedef tpl_object#(int, "tpl_object_2") tpl_object_2;
typedef tpl_object#(int, "tpl_object_3") tpl_object_3;
typedef tpl_object#(int, "tpl_object_4") tpl_object_4;
typedef tpl_object#(int, "tpl_object_5") tpl_object_5;
typedef tpl_object#(int, "tpl_object_6") tpl_object_6;
typedef tpl_object#(int, "tpl_object_7") tpl_object_7;
typedef tpl_object#(int, "tpl_object_8") tpl_object_8;
typedef tpl_object#(int, "tpl_object_9") tpl_object_9;
typedef tpl_object#(int, "tpl_object_10") tpl_object_10;
typedef tpl_object#(int, "tpl_object_11") tpl_object_11;
typedef tpl_object#(int, "tpl_object_12") tpl_object_12;
typedef tpl_object#(int, "tpl_object_13") tpl_object_13;
typedef tpl_object#(int, "tpl_object_14") tpl_object_14;
typedef tpl_object#(int, "tpl_object_15") tpl_object_15;
typedef tpl_object#(int, "tpl_object_16") tpl_object_16;
typedef tpl_object#(int, "tpl_object_17") tpl_object_17;
typedef tpl_object#(int, "tpl_object_18") tpl_object_18;
typedef tpl_object#(int, "tpl_object_19") tpl_object_19;
typedef tpl_object#(int, "tpl_object_20") tpl_object_20;
typedef tpl_object#(int, "tpl_object_21") tpl_object_21;
typedef tpl_object#(int, "tpl_object_22") tpl_object_22;
typedef tpl_object#(int, "tpl_object_23") tpl_object_23;
typedef tpl_object#(int, "tpl_object_24") tpl_object_24;
typedef tpl_object#(int, "tpl_object_25") tpl_object_25;
typedef tpl_object#(int, "tpl_object_26") tpl_object_26;
typedef tpl_object#(int, "tpl_object_27") tpl_object_27;
typedef tpl_object#(int, "tpl_object_28") tpl_object_28;
typedef tpl_object#(int, "tpl_object_29") tpl_object_29;

endpackage
```

```SystemVerilog
//top.sv
module top ();
    task hello();
        $display("Hello World!");
    endtask

    export "DPI-C" task hello;
    import "DPI-C" context function void hello_wrap();

    initial begin
        hello_wrap();
        $finish;
    end
endmodule
```

Command to reproduce:
- ubuntu
```bash
g++ -O3 -Wall -shared -std=c++17 -fPIC py_dpi.cpp -o py_dpi.so -I /usr/local/share/verilator/include/vltstd
verilator -j $(nproc) --binary -top top tpl.sv tpl_top.sv $PWD/py_dpi.so 
./obj_dir/Vtop
```
- macOS
```bash
g++ -O3 -Wall -shared -std=c++17 -fPIC  -undefined dynamic_lookup py_dpi.cpp -o py_dpi.so -I /opt/homebrew/Cellar/verilator/5.028/share/verilator/include/vltstd
verilator -j $(nproc) --binary -top top tpl.sv tpl_top.sv $PWD/py_dpi.so 
./obj_dir/Vtop
```

wsnyder, 2024-10-07:
The attached, which I think matches what you describe, works ok for me using master on Ubuntu 22.04.
[bug5513.tgz.txt](https://github.com/user-attachments/files/17285167/bug5513.tgz.txt)

wsnyder, 2024-11-02:
Did this resolve for you?  If not please update the example in last comment.

Dragon-Git, 2024-11-02:
Hi wsnyder,

Thank you for your continued support and assistance.

I have updated to Verilator 5.030 and tested the example you provided. Unfortunately, I am still encountering the same linking errors. Here are the details of my environment and the steps I took:

Environment Details:

- Operating Systems:
  - Ubuntu 20.04.6 LTS
  - macOS Sequoia 15.0.1
- Verilator Version: Verilator 5.030 2024-10-27 rev v5.030
- Compilers:
  - Ubuntu: gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
  - macOS: Apple clang version 16.0.0 (clang-1600.0.26.3)

Steps Taken:
- Updated Verilator to version 5.030.
- Compiled the provided example code on both Ubuntu and macOS.
- Added the -undefined dynamic_lookup flag to the macOS g++ options.

Error Message:
```bash
# on ubuntu
> g++    verilated.o verilated_dpi.o verilated_threads.o Vtop__ALL.a   /workspace/bug5513/py_dpi.so  -pthread -lpthread -latomic   -o Vtop
/usr/bin/ld: /workspace/bug5513/py_dpi.so: undefined reference to `hello'
collect2: error: ld returned 1 exit status
make[1]: *** [Vtop.mk:66: Vtop] Error 1
```

```bash
# on macOS
> ./obj_dir/Vtop
dyld[3307]: symbol not found in flat namespace '_hello'
make: *** [x] Abort trap: 6
```

wsnyder, 2024-11-02:
Does the example I sent do that on Ubuntu 20?

What you are doing is unusual/unique and probably not an issue in Verilator itself, so generally you'll need to do the debug debug.  Perhaps adjust the order of flags on the link line?

Dragon-Git, 2024-11-04:
The example you sent doesn't work successfully on Ubuntu 20.

I have managed to resolve the linking issues by modifying the Makefile generated by Verilator. Specifically, I added the `-Wl,--whole-archive` and `-Wl,--no-whole-archive` options to the link rules in the `Vtop.mk` file. This approach worked successfully on Ubuntu 20.04.6 LTS with gcc 9.4.0. Here are the details:

### Environment Details:
- **Operating System:** Ubuntu 20.04.6 LTS
- **Verilator Version:** Verilator 5.030 2024-10-27 rev v5.030
- **Compiler:** gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)

### Modified Makefile (`Vtop.mk`):
```makefile
### Executable rules... (from --exe)
VPATH += $(VM_USER_DIR)

### Link rules... (from --exe)
Vtop: $(VK_USER_OBJS) $(VK_GLOBAL_OBJS) $(VM_PREFIX)__ALL.a $(VM_HIER_LIBS)
	$(LINK) $(LDFLAGS) -Wl,--whole-archive $^ -Wl,--no-whole-archive $(LOADLIBES) $(LDLIBS) $(LIBS) $(SC_LIBS) -o $@
```

### Explanation:
- **`-Wl,--whole-archive`**: This option ensures that all symbols from the specified archives are included in the final executable, even if they are not directly referenced.
- **`-Wl,--no-whole-archive`**: This option resets the behavior to the default, where only referenced symbols are included.

By adding these options, the linking process correctly includes all necessary symbols, resolving the undefined symbol errors.

But clang doesn't have that option, I'll look for a solution for clang (macOS).

wsnyder, 2024-11-04:
Great, this will help the next person needing this.  Please keep us updated. The flags needed are a consequence of the dynamic linking etc in your use case. As this is not a generally applicable problem we won't need/want  to change the default linker flags.

closed 2024-11-04 by wsnyder