verilator/verilator #5517: Argument list too long

Dragon-Git, 2024-10-07:
When I build with `-MAKEFLAGS VM_PARALLEL_BUILDS=0` make flag, the Verilator run will fail with the following error message.
```
make: execvp: /bin/sh: Argument list too long
make: *** [/usr/local/share/verilator/include/verilated.mk:209: Vdbg_tpl_pkg__ALL.cpp] Error 127
make: *** Waiting for unfinished jobs....
make: Leaving directory '/workspace/uvm_debug/tpl_bug/obj_dir'
%Error: make -C obj_dir -f Vdbg_tpl_pkg.mk -j 16 VM_PARALLEL_BUILDS=0 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/share/verilator/bin/verilator_bin -j 16 --binary -top dbg_tpl_pkg -MAKEFLAGS VM_PARALLEL_BUILDS=0 tpl.sv
```

- Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```systemverilog
//tpl.sv
package dbg_tpl_pkg;
    class tpl_object#(type T=int, string NAME = "");
        T t;
        function new(string name = NAME);
            this.t = $urandom_range(0, 100);
        endfunction

    endclass

// from 0 to 399
typedef tpl_object#(int, "tpl_object_0") tpl_object_0;
typedef tpl_object#(int, "tpl_object_1") tpl_object_1;
typedef tpl_object#(int, "tpl_object_2") tpl_object_2;
typedef tpl_object#(int, "tpl_object_3") tpl_object_3;
...
...
...
typedef tpl_object#(int, "tpl_object_398") tpl_object_398;
typedef tpl_object#(int, "tpl_object_399") tpl_object_399;

endpackage
```

- What command line do we use to run your example?

```bash
verilator -j $(nproc) --binary -top dbg_tpl_pkg -MAKEFLAGS VM_PARALLEL_BUILDS=0 tpl.sv
```

- What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.026 2024-06-15 rev v5.026

- What OS and distribution are you using?

Ubuntu 20.04.6 LTS

wsnyder, 2024-10-07:
Not surprising.  Fix is to build the list in the Makefile using files, which will need many tries to get right.  Or Verilator can emit the concatenated file directly.

However, master has a new feature, `--output-groups` which will limit the file count and likely mean this number of files won't be relevant any longer.  Can you see if that fits your larger use case?

Dragon-Git, 2024-10-08:
Thank you for the suggestion and for pointing out the `--output-groups` option. I have tried it, and it successfully addresses my issue. This new feature is a great addition and fits well with my use case.

wsnyder, 2024-11-05:
Closing for now as unlikely to get attention, though if this starts popping up more a pull would be accepted.  Recommendation is to use `--output-groups`, which might become default once stable/proven.