verilator/verilator #5521: Run Different Models on Different Threads

alpha-beta-user, 2024-10-08:
This is a gerat job!

I want to know If I can run Different models on different Threads/cores. My RTL model is a large project. Then I translate the rtl project into systmec model. 

But if I simulate the whole systemc project, it is time-consuming. It takes too much time to run over.

So I want to know if any methods to accelarated the simulated speed, maybe run parallel on different threads/cores. Maybe Varitor support generate such multi-thread systemc model.

Thanks for your attention.

wsnyder, 2024-10-08:
Please see the documentation and notes on --sc use: https://veripool.org/guide/latest/verilating.html#multithreading

If after that you have additional questions feel free to follow up here.

closed 2024-10-08 by wsnyder

alpha-beta-user, 2024-10-10:
I find that if I set threads 4, it is much slower than single thread. even I make the design more complex.

can you tell me reason.

thanks for your answer!

wsnyder, 2024-10-10:
Probably the design is still very small. Run thread profiling as described in the manual to see if/why.

wsnyder, 2024-10-10:
As a general metric, anything faster than ~~10k clocks/second is small and might not benefit from threading.

alpha-beta-user, 2024-10-10:
> As a general metric, anything faster than ~~10k clocks/second is small and might not benefit from threading.

do you mean the clock cycle should be laeger enough?

1 / 10000 = 10^(-4)s = 10^5 ns

clock cycle should be bigger than 10^5ns

Is It right?

thank you for your reply?

wsnyder, 2024-10-10:
I mean the performance of the model relative to wall time.  I thought we printed this in the gantt report, but I guess not, you can see it in the summary:

```
- S i m u l a t i o n   R e p o r t: Verilator 5.029 devel
- Verilator: $finish at 17ps; walltime 10.005 s; speed 2.841 ns/s
```

So if your modeled clock is 1ns, this is running 2.84 clocks/second.  Make sure you simulate for many wall-seconds for this or any profile to be accurate.

alpha-beta-user, 2024-10-10:
> I mean the performance of the model relative to wall time. I thought we printed this in the gantt report, but I guess not, you can see it in the summary:
>
> ```
> - S i m u l a t i o n   R e p o r t: [...]

Thank you for answer!

I want to know how can I get the report.

I read the verilator --help, can not get it.

Thank you!

wsnyder, 2024-10-10:
https://veripool.org/guide/latest/simulating.html#simulation-summary-report