verilator/verilator #5524: @(posedge some_variable_frq_gated) triggered, even when some_variable_frq_gated didn't changed

Febbe, 2024-10-09:
I wrote a stepper motor position driver, which triggers on a gated clk with variable frequency.
Testing the module directly yields the expected result, but as soon, I test it with a gated clock (additionally with a frequency change), the module triggers on a not available posedge.

### Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
I only can show the module where it fails and a direct testcase, the code, generating the gated clock with variable frequency can't be shared.

```
// Copyright (c) 2024 Fabian Keßler

`default_nettype none
`include "autofokus_ports.svh"
import af::*;

module mk_stepper_motor_position (
  input  var logic        clk_control,          // Motor step trigger
  input  var logic        rst_n,                // Active-low reset signal
  input  var logic        i_direction,          // Direction signal
  input  var logic        i_reset_zero,         // Reset position to zero
  output var logic        o_reset_zero_acq,
  output var logic [17:0] o_current_position    // Current position
);

position_t current_position_reg, next_pos;

  always_ff @(posedge i_reset_zero, negedge i_reset_zero)
    o_reset_zero_acq <= i_reset_zero;

  always_ff @(posedge clk_control, negedge rst_n, posedge i_reset_zero) begin
    if (!rst_n || i_reset_zero) begin
      // Asynchronous Reset: Initialize on reset
      current_position_reg <= 18'h0;
    end else begin
      current_position_reg <= current_position_reg + ((i_direction != 0) ? -1: 1);
    end
  end

  assign o_current_position = current_position_reg;

endmodule

```

```
// Copyright (c) 2024 Fabian Keßler

module tb_mk_stepper_motor_position;
  parameter CLK_PERIOD = 2;  // Clock period in ns

  // Signal declarations
  logic clk;                // Clock signal
  logic gated;
  logic enable;
  logic reset;              // Reset signal
  logic [17:0] position;    // Output position signal

  // Instantiate the mk_steper_motor_position module
  mk_stepper_motor_position uut (
    .clk_control(gated),
    .rst_n(reset),
    .i_direction(0),
    .i_reset_zero(0),
    .o_reset_zero_acq(),
    .o_current_position(position)
  );

  assign gated = enable? clk: 0; 

  initial begin
    clk = 0;
    forever #(CLK_PERIOD / 2) clk = ~clk;  // Clock toggles every half-period
  end

  initial begin
    $dumpfile("tb_mk_stepper_moter_position.vcd");
    $dumpvars;

    enable = 0;
    reset = 1;
  
    #20 reset = 0;
    #20 reset = 1;

    #(CLK_PERIOD * 3) enable = 1;
    #(CLK_PERIOD * 3) enable = 0;
    #(CLK_PERIOD * 3);
    #(CLK_PERIOD * 3);

    $finish;
  end

endmodule
```
Working waveform:
[image]

Same module tested with gated variable frq:
[image]

### What 'verilator' command line do we use to run your example?
cmake:
```
verilate(target SOURCES <tb_toplevel_name>.sv <mk_module>.sv VERILATOR_ARGS --timing --main --trace --exe)
```

### What 'verilator --version' are you using?  Did you try it with the git master version?
5.29
What OS and distribution are you using?

May we assist you in trying to fix this in Verilator yourself?
Yes.

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-10-09:
Most likely there's a glitch that resolves itself so is not visible on waves, for example you are changing `enable` at the same time as `clk` in your level above.  As you didn't share that code I can't say for sure.

Febbe, 2024-10-09:
> Most likely there's a glitch that resolves itself so is not visible on waves, for example you are changing `enable` at the same time as `clk` in your level above. As you didn't share that code I can't say for sure.

Indeed, the same clock which drives the gated clock also drives the enable signal. This might cause a metastable state, which is not represented in the waveform. Would it be possible, to add such cases in the waveform, instead of silently resolving them?

wsnyder, 2024-10-09:
Some of the commercial tools offer "delta tracing" I would use one of those if you need this.  We're not going to add that as it will slow down tracing and normal simulations, plus there's no way to represent it with the waveform formats we support (AFAIK, but perhaps some hack is possible).

closed 2024-10-09 by wsnyder